prusa3d/PrusaSlicer #5662: Full fan speed at layer does not ramp up

Funkton, 2021-01-02:
### Version
2.3.0-rc2+win64

### Operating system type + version
Windows 10 Pro for workstations
10.0.18363

### 3D printer brand / version + firmware version (if known)
Modified Qidi X-Max

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
The preview text for the fan speed does not show the full fan speed after full fan speed layer. Instead it stays at the minimum fan speed. gcode slices confirm.
Expected behavior would be to ramp up the fan to maximum by the layer in the setting.

[image]

rtyr, 2021-01-03:
It doesn't work like that. Full fan speed doesn't mean 100% fan speed. Full fan speed depends on layer time, so it can be value between 35-100% in this case. 

[image]

Funkton, 2021-01-03:
Ok I see. I was confused on the function based on the hover tooltip. Maybe some adjustment to the text would be helpful.
I was under the assumption that the "Full fan speed at layer" setting would use the maximum setting and not the minimum.
What does this setting actually do then? The tooltip states it ramps up to maximum, not minimum.
[image]

rtyr, 2021-01-04:
You can set the number of steps (layers) to achieve the regular/final/full fan speed more smoothly. It can prevent temperature drops when the fan kicks in. It is mainly meant for initial layers of the print.

Simple example:
Layer1: 0% fan speed, Layer2: 100% fan speed (=possible temperature drop)
[image]

Layer1: 0% fan speed, Layer2: 50%, Layer3: 100%
[image]

Funkton, 2021-01-04:
Ok, I suggest the tooltip text be updated to indicate actual functionality. Instead of saying it ramps up to maximum, that it ramps to minimum.

rtyr, 2021-01-04:
No, it doesn't ramp up to the minimum.
But maybe te tooltip can be improved.

Closing.

closed 2021-01-04 by rtyr