prusa3d/PrusaSlicer #5668: Issue slicing part originally designed with OpenScad

pellcorp, 2021-01-03:
### Version
2.3.0rc2

### Operating system type + version
Ubuntu 20.10

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 + Latest Marlin 1.1 firmware

### Behavior
Slicing a part adds layers where there were none in the original part.

Add STL, and change the orientation so its flat on the bed:

[image]

Slice

[image]

* _Expected Results_

Should slice the file correctly

* _Actual Results_

As above, did not slice the file correctly

The STL was downloaded from:

https://www.thingiverse.com/thing:1264567

I tried re-exporting it with OpenScad, no change.   I tested slicing this in PrusaSlicer 2.2.0 same issue as with 2.3.0rc2.  With Cura it sliced fine, but I much prefer PrusaSlicer so Cura is not an option!

rtyr, 2021-01-03:
Repair the model, it will most likely help.

pellcorp, 2021-01-03:
I already tried that as well, same result.

rtyr, 2021-01-03:
So this STL does not work for you?

[GuessWho_repaired.zip](https://github.com/prusa3d/PrusaSlicer/files/5762308/GuessWho_repaired.zip)

pellcorp, 2021-01-04:
It did, what tool did you use to repair it?  I used PrusaSlicer to repair and then added the resulting object back in, and there was no change.

rtyr, 2021-01-04:
Classic Netfabb repair from PrusaSlicer (not available in Linux version).

pellcorp, 2021-01-04:
Ok, so not helpful for me I guess.   Cura managed to import the model just fine, so I think its still a deficite in PrusaSlicer that it failed to handle it, but it sounds like perhaps OpenScad is as much to blame.

So Im assuming this report will probably go nowhere, in which case I will just close the issue.

Thanks for your help

closed 2021-01-04 by pellcorp

JohnOCFII, 2021-01-05:
You can try the free online web-based NetFabb repair tool.  I use it regularly.  https://service.netfabb.com/login.php

pellcorp, 2021-01-05:
Thanks JohnOCFII, would be cool if PrusaSlicer integrated the web netfabb service for OS that don't have support for the NetFabb OS integration.

But again, Im assuming this is not really upsetting anyone so unlikely that anything happens here.    I will keep it in mind though for the future.

pellcorp, 2021-01-05:
Hmm, online netfabb did not like it either!

[image]

pellcorp, 2021-01-05:
However 3D Builder from a Windows 10 VM I have fixed it fine!

Amazing tool 3d Builder

bubnikv, 2021-01-05:
@pellcorp Then PrusaSlicer running from the same Windows 10 VM would fix it exactly the same way. Indeed, PrusaSlicer on Windows 10 calls the Windows 10 API for fixing the model the same way 3D Builder does.

pellcorp, 2021-01-05:
@bubnikv, thats good to know, but I don't use Windows 10 for 3d printing, I much prefer linux.    So I will put this down to a dodgy OpenScad model, and move on.  Thanks for all the responses, I really do love PrusaSlicer, just a shame that Cura managed the model fine as is without repair, but PrusaSlicer did not.   But I can now use 3d Builder to repair the model if I encounter this issue again.   

I wonder if there is a way to get this to work from wine, so I don't have to start the VM :-)  Something for another day.

bubnikv, 2021-01-05:
Wine will not help, you need Windows 10 services for that.

pellcorp, 2021-01-05:
Oh well, I can dream!  :-)

FidelCapo, 2021-02-01:
This issue should be corrected in next release without the need to repair the model using the netfabb service.

bubnikv, 2021-07-02:
Released with PrusaSlicer 2.3.2-beta.