prusa3d/PrusaSlicer #5669: overhang angle is not respected

SteWhite, 2021-01-03:
### Version
2.3.0-rc2
### Operating system type + version
Windows 10 pro 64bit
### 3D printer brand / version + firmware version (if known)
MK3S
### Behavior
The angle I set(45°) is not respected.
You can see the different behavior of Simplify3D, cura, ideamaker
[image]
[image]

_Is this a new feature request?_
no

rtyr, 2021-01-04:
Please provide your 3MF project file.

SteWhite, 2021-01-04:
#### Project File (.3MF) where problem occurs
[svuotatasche.zip](https://github.com/prusa3d/PrusaSlicer/files/5766994/svuotatasche.zip)

closed 2026-09-29 by prusaslicer-bot[bot]