prusa3d/PrusaSlicer #5670: paint seams lines show as dots on opposite side

paloss, 2021-01-03:
### Version
2.3.0 rc 2 win 64

### Operating system type + version
win 10 home 64bit 

### 3D printer brand / version + firmware version (if known)
mk3s - 3.9.0-3421

### Behavior
When painting seams lines on for example the back the seams also show up as dots on the opposite side. This only seems to happen on heavy scaled objects

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
[image]
[image]

lukasmatena, 2021-01-05:
I managed to reproduce it with another heavily scaled object. It looks like the problem is not the new painter, but probably the old raycaster. I reproduced the same problem when adding support points in SLA mode in 2.2.0.

neophyl, 2021-01-05:
I’ve also seen this with painting support enforcers on to models without them being scaled too. I sometimes get blue sections on the other side of the model even though they shouldn’t be there. For me it’s not been a problem as the areas it added them too weren’t getting supports anyway (on top of the model)

lukasmatena, 2021-01-05:
@neophyl Do you have a 3mf?

neophyl, 2021-01-05:
Here you go, should be a 3mf file attached as a zip.  The support enforcers were painted on the bottom from below the part and 'bled through' to the top.  The original 3mf had other parts on there but this was the part that I noticed had the issue.   As I say it didnt actually cause me a problem as the support was only generated on the bottom but I can understand how its looks like its possible for the painting to appear in other areas than you expect.  

[MandoBerg_v7_support painting_bleed_through.zip](https://github.com/prusa3d/PrusaSlicer/files/5771977/MandoBerg_v7_support.painting_bleed_through.zip)

lukasmatena, 2021-01-06:
@neophyl Thanks. I think that in your case it does not paint through the model, but that the color bleeds through the holes - which is exactly what it should do when you use the 'Sphere' brush type. Circle brush does not do it. Could that be so?

neophyl, 2021-01-06:
Quite possibly, to be honest I hadn't noticed the different brush types :)  I'll be sure to switch to circle in future and try that.

bubnikv, 2021-09-22:
We don't see any issue, at least with PrusaSlicer 2.4.0-alpha1

@paloss would you please attach your 3MF?

paloss, 2021-09-23:
Had to put it on wetranfser for some reason. 

i resized the model to be 175mm height. thats a ridiculous amount of 20951% enlargement. 

stl file: https://we.tl/t-GowZnKCpno

closed 2026-09-29 by prusaslicer-bot[bot]