prusa3d/PrusaSlicer #5671: Don't prompt to save profile is current project 3mf file has been saved

nordurljosahvida, 2021-01-03:
### Version

2.3.0-rc2 GTK3 appimage

### Operating system type + version

Ubuntu 20

### Behavior

* _Describe the problem_

Even if a project is saved, upon quitting PS prompts to save all changes into a profile.

* _Steps needed to reproduce the problem_

Open a slicing session, make changes to print/printer/filament, save the 3mf, and quit

* _Expected Results_

For such a modular and flexible program like PS, there is no reason - at least for us [3d printing shop] - to save profiles in PS. We save 3mfs for every project, and all settings are re-applied any time we reopen that project. I understand that profiles might be a good option for those doing smaller projects and perhaps not versioning every file of every project in its own 3mf file, but *if* a 3mf file was saved, there is no reason to prompt to save or "lose" changes, because those changes would never be saved.

I even got a live chat rep [who was, as always, otherwise very skilled and knowledgeable] to tell me that if I don't save something into a profile, that won't be applied to the exported gcode, whereas that is obviously false as I've literally never saved a single print or filament profile in my entire life.

I know there's a "remember this decision" checkbox, but we should definitely distinguish between "3mf has been saved" and "3mf has not been saved or further changes to the profiles have been applied since last save" scenarios, since, right now, if I checked the "remember this decision" box, and then pressed "discard" I'd be at risk of losing actually *unsaved* work in the future. 

* _Actual Results_

PS asks to save changes to a profile regardless of whether a 3mf was saved or not

_Is this a new feature request?_

yes, thanks in advance

Area5142, 2021-01-03:
I find the way PrusaSlicer handles changes confusing too.

A PS project should be handles the same way as a document in a word-processor this would be intuitive for the majority of users.

bubnikv, 2021-01-04:
You are guys right. We will get there one day. The current behavior is a
result of multiple refactorings of the old Slic3r.

nordurljosahvida, 2021-01-04:
I do love the new discard / save / transfer behaviour, that is really great. This is only a small extra to make it perfect in my opinion ;]

lukasmatena, 2021-08-23:
This is now implemented in 2.4.0-alpha1. In case current project is saved, the "Unsaved changes" dialog for the profiles is not shown. It is only shown when it is unsaved and user decides to discard the changes.

I'm closing the issue. We'll be glad if you provided some feedback after the alpha is released (which should be shortly). Thanks.

closed 2021-08-23 by lukasmatena

nordurljosahvida, 2021-08-23:
Very good, thank you!