prusa3d/PrusaSlicer #5673: Memory bad allocation

Mikeant, 2021-01-03:
### Version
2.3.0 rc2 32 bit, version 2.2.0 generating error too

### Operating system type + version
Win 7 Pro 32 bit
4 GB RAM

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro

### Behavior
In bigger models PrusaSlicer crash while Slicing in phase Generate G-code
Error Bad allocation
How many memory I need in 32 bit system?

Název události problému:	APPCRASH
  Název aplikace:	prusa-slicer.exe
  Verze aplikace:	2.3.0.0
  Časové razítko aplikace:	5fe21ff5
  Název chybného modulu:	ucrtbase.DLL
  Verze chybného modulu:	10.0.10240.16390
  Časové razítko chybného modulu:	55a5bf73
  Kód výjimky:	40000015
  Posun výjimky:	0007d85a
  Verze operačního systému:	6.1.7601.2.1.0.256.48
  ID národního prostředí:	1029
  Další informace 1:	9594
  Další informace 2:	959442177260a16e72eb20823e30ce55
  Další informace 3:	c20f
  Další informace 4:	c20f06e07735d2a72186daed01618ca2

#### Project File (.3MF) where problem occurs
[Cat.zip](https://github.com/prusa3d/PrusaSlicer/files/5761789/Cat.zip)

bubnikv, 2021-01-04:
Most likely you have insufficient RAM or video RAM.

rtyr, 2021-09-21:
Answered. No response. Closing.

closed 2021-09-21 by rtyr