prusa3d/PrusaSlicer #5676: Prusa 2.30-RC2 wont open mapped network drives (cant see them in file->open)

wlcina, 2021-01-03:
### Version
_Version of PrusaSlicer used goes here_
Version 2.3.0-rc2+win64

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10 64bit

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
MK3S+

### Behavior
Prusa Slicer cant open network drives (mapped network drives) when opening new file - FILE>Open, network drives are not listed. Also drag and drop from network drive is not working. Prusa 2.20 support network drives and drag and drop on the same machine.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Prusa 2.30 RC2
[image]

Prusa 2.20 (running simultaneously on the same machine)
[image]

wlcina, 2021-01-03:
Update - drag and drop wont work at all :-(

foreachthing, 2021-01-03:
Do you start "Prusa 2.30 RC2" as admin?

bubnikv, 2021-01-04:
I cannot reproduce your issue. I can open a 3MF from a network drive both by File->Open Project and by Drag & Drop.

wlcina, 2021-01-04:
Running PS 2.30 RC2 as admin, restarted computer - still no network drive can be opened (they are not listed), from what I know, the file requestor can be set to wont allow to open network drives (eg win32img has it set this way), strange is that 2.20 opens network drives. 
(No project can be opened or STL from network drive, or any file, network drives are just not listed)

bubnikv, 2021-01-04:
Three other developers of our team were not able to reproduce your issue either.

wlcina, 2021-01-04:
I will try to reproduce problem on another computer with windows installation - same issue appears, I did some research, and I found I am missing registry value named "EnableLinkedConnections", I created one (steps are bellow) and it works now. I am not sure why it worked with 2.20 without this registry value. 

Drag and drop issue still persist, when trying to drag and drop STL file mouse cursor changes to circle with diagonal line (like no entry sign) and wont allow me drag and drop anything. I found there are more people with the same issue (before):
issue #2212
issue #5035 

Drag and drop is working on 2.20, never met this issue before.

--[Steps to enable network drive access for PrusaSlicer 2.30RC2]---------------------------------------
Click Start, type regedit in the Start programs and files box, and then press ENTER.
Locate and then right-click the registry subkey HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\CurrentVersion\Policies\System.
Point to New, and then click DWORD Value.
Type EnableLinkedConnections, and then press ENTER.
Right-click EnableLinkedConnections, and then click Modify.
In the Value data box, type 1, and then click OK.
Exit Registry Editor, and then restart the computer.

wlcina, 2021-01-04:
Fix for drag and drop is also in the registry, just tested it and this fix works.
Fix needs to disable LUA, may be dangerous for some users.
I think it should be investigated further (maybe try to check if devs have it enabled or disabled first)

--[Steps to enable drag and drop feature in PrusaSlicer 2.30RC2]---------------------------------------
Click Start, type regedit in the Start programs and files box, and then press ENTER.
This will open Windows Registry Editor.
Navigate to: HKEY_LOCAL_MACHINE > SOFTWARE > Microsoft > Windows > CurrentVersion > Policies > System
Find the key EnableLUA, and change 1 to 0.
Exit Registry Editor, and then restart the computer.

bubnikv, 2021-01-04:
I suppose we developers are all admins of the machine. Maybe that makes a difference?

wlcina, 2021-01-04:
> I suppose we developers are all admins of the machine. Maybe that makes a difference?

I believe it could be something with Windows 10 update, however fixes are working, so this can be closed and archived if anyone else meet the issue.

foreachthing, 2021-01-04:
Reconnect your network drives as admin. 
Maybe your admin has no permission.

wlcina, 2021-01-15:
Fixes works on new Prusaslicer 2.30

closed 2021-01-15 by wlcina

bubnikv, 2021-01-15:
The difference may be, that the 2.3.0 release installer is signed, while the binary releases on github are not.