prusa3d/PrusaSlicer #5678: Feature Request - separate Ironing hot end temp

THiNK3dAUS, 2021-01-04:
**Version**
Prusa Slicer 2.3 Release Candidate 2

**Operating system type + version**
Windows 10 64bit

**3D printer brand / version + firmware version (if known)**
Prusa MK3S - FW 2.9.0
(additional info - Bear EXXA Extruder, Phaetus Dragon Hi-flow hotend, e3D NozzleX 0.4mm)

**Behavior**
Have been experiencing bad clogging when utilising the ironing feature and PLA materials, as described in the 'downsides' section of the knowledge base. This appears related to heat creep, which then stops the material extruding, and ironing reverting to effectively scrapping a hot nozzle across the part. 

Following some testing across numerous prints, it appears that whilst adjustment to the current included settings within PS2.3rc2 can assist (specifically line width and speed), the main adjustment that appears to have alleviated the issue is increasing the hot end temperature to help keep the material flowing. 

Currently, this either needs to be done manually on the fly, or by adjusting the g-code file after slicing. 

Adding an additional option with PS to adjust the hot end temp when ironing is started would be a good solution.
Suggestion would be to be included under the "filament" settings, to allow for filament specific settings (much like 'first layer' temps).

R0dri, 2021-03-18:
Wouldn't it be better reduce the temperature while ironing if the ironing time exceeds a certain threshold? I found there are no issues when the layer being ironed is not too big.

kaetemi, 2021-03-18:
+1

When I use a stringy PLA (eSun PLA+), ironing worked well for me so far.

Using a more consistent PLA (PopBit T-PLA), the PLA gums up in the nozzle at 200°C while ironing.

Increasing temperature to 220°C and a little push-pull action got the PLA flowing again and saved the rest of the print. Ironing stayed reliable at this temperature.

kaetemi, 2021-03-18:
Some further testing. Ironing at higher temperature clogged up too. Punching up the temperature again let me unclog it easily. Dropping the temperature down to 180° after unclogging worked fine, so far.

dylan546378, 2021-09-28:
+1, my nozzle will scrape on excess filament on the z axis while ironing in some cases. I get better results when I manually raise the temp while ironing

andyylin, 2022-10-29:
I've just also run into the clogging after ironing issue as well with a 0.6mm nozzle. I wish PrusaSlicer had a way of setting different temperatures during the print.

neophyl, 2022-10-29:
Use the post processing function that is built in to PS now. It is what it is there for.

audkar, 2026-09-04:
Still relevant