prusa3d/PrusaSlicer #5679: How to orbit when drag starts ON the model?

soswow, 2021-01-04:
### Version
2.3.0-rc1+

### Operating system type + version
Mac

### Behavior
Pan works with right mouse button and surprisingly with middle button as well.

Orbit works with left mouse button, BUT! only when drag starts over non-model. I often zoomed it very closely, and there is no way to find non-model in the view to start orbiting. Is there a way to do this?

_Partially a feature request_
Also, is center of rotation constant? It would be awesome for it to be where mouse touching a model, like it is in Fusion 360.

bubnikv, 2021-01-04:
> Orbit works with left mouse button, BUT! only when drag starts over
non-model. I often zoomed it very closely, and there is no way to find
non-model in the view to start orbiting. Is there a way to do this?

Only in PrusaSlicer 2.3 and if one is in a Gizmo mode (Cut tool active,
support painting active etc). Then holding the Control key will allow you
to rotate or pan even if the mouse is over an object.

soswow, 2021-01-04:
yes. This is what I was looking for. I was testing before tools were activated 🤦🏻‍♂️.
How about I close this issue and open a feature request described in the last section of my issue description? About center of rotation.

soswow, 2021-01-04:
Actually ... center of rotation seems to be relative to the camera position and zoom level. It is not bad already. But if it could be related to where mouse is (ray touching a model from a camera through the mouse cursor) that would make it awesome!

bubnikv, 2021-01-05:
> But if it could be related to where mouse is (ray touching a model from a
camera through the mouse cursor) that would make it awesome!

I have implemented this behavior some 20 years ago and the experience was
not pleasant either. The center of rotation changes with each mouse down
event. It feels weird. I understand the utility, but the current behavior
is more intuitive.

soswow, 2021-01-05:
Interesting. Here is how it works in Fusion 360. Personally this is most intuitive way out there. One important feature they have - is to show the center of rotation that is chosen. Maybe this is a missing feature. Here is a demo:

https://user-images.githubusercontent.com/108065/103621845-ef5e9880-4f89-11eb-88ab-b44c75d5e6e5.mp4

bubnikv, 2021-01-05:
It is true that I may not have shown the center of rotation at that time.
We may try to implement the Fusion360 way, it is not technically difficult.

ristein, 2021-11-04:
I like the movements in systems like Catia, where one can select the Center of Rotation with a middle mouse click.
So anyone not liking this feature can just ignore the middle mouse click but others can easily set the CoR at the feature they want to inspect.

closed 2026-09-29 by prusaslicer-bot[bot]