prusa3d/PrusaSlicer #5683: Feature request: "--export-stl-with-supports"

jtoberling, 2021-01-04:
### Version
2.2.0+linux_x64

### Operating system type + version
Linux Mint 20 Ulyana

### 3D printer brand / version + firmware version (if known)
Proprietary / Custom

### Behavior

Could you please add command line option (eg:--export-stl-with-supports )  to the functionality provided by the menu point:
   "File / Export / Export plate as STL including supports ..." ?

_Is this a new feature request?_
Yes

Gregregorovich, 2023-03-02:
This appears to be implemented for "SLA supports" as of May 2019 (see https://github.com/prusa3d/PrusaSlicer/issues/1999 ), but not for FDM mode or whatever it's called. With custom painted (or even auto-generated painted) supports being a thing, this would be especially useful now.

k-gordon, 2023-03-02:
> This appears to be implemented for "SLA supports" as of May 2019 (see #1999 ), but not for FDM mode or whatever it's called. With custom painted (or even auto-generated painted) supports being a thing, this would be especially useful now.

Hi, this feature request is specifically for a command line request, which is not implemented as of yet.

hradec, 2024-03-20:
I'm also waiting for this. I was surprised today that the `--export-stl` or `--export-obj` command line parameters doesn't trigger the support generation and doesn't export it.

hradec, 2024-03-21:
I was able to hack it today by doing this: 
```
diff --git a/src/PrusaSlicer.cpp b/src/PrusaSlicer.cpp
index 79bb8b6f9..c41c3a880 100644
--- a/src/PrusaSlicer.cpp
+++ b/src/PrusaSlicer.cpp
@@ -628,7 +628,7 @@ int CLI::run(int argc, char **argv)
                 }
                 if (print->empty())
                     boost::nowide::cout << "Nothing to print for " << outfile << " . Either the print is empty or no object is fully inside the print volume." << std::endl;
-                else
+                else {
                     try {
                         std::string outfile_final;
                         print->process();
@@ -656,6 +656,23 @@ int CLI::run(int argc, char **argv)
                         boost::nowide::cerr << ex.what() << std::endl;
                         return 1;
                     }
+
+                    // =======================================================================================================================================
+                    // export the mesh + support to a file
+                    // =======================================================================================================================================
+                    SLAPrint *slaprint = static_cast<SLAPrint*>(print);
+                    TriangleMesh *mesh = ( TriangleMesh * )&(slaprint->objects()[0]->support_mesh());
+                    mesh->merge(slaprint->objects()[0]->pad_mesh());
+
+                    std::shared_ptr<const indexed_triangle_set> m = slaprint->objects()[0]->get_mesh_to_print();
+                    TriangleMesh inst_object_mesh;
+                    if (m)
+                        inst_object_mesh = TriangleMesh{*m};
+                    mesh->merge(inst_object_mesh);
+                    Slic3r::store_obj("/tmp/xxx.obj", mesh);
+                    // =======================================================================================================================================
+                }
+
 /*
                 print.center = ! m_config.has("center")
                     && ! m_config.has("align_xy")
diff --git a/src/libslic3r/SLAPrint.hpp b/src/libslic3r/SLAPrint.hpp
index 4148745f8..0bccee980 100644
--- a/src/libslic3r/SLAPrint.hpp
+++ b/src/libslic3r/SLAPrint.hpp
@@ -553,6 +553,7 @@ public:
                       const ThumbnailsList &thumbnails,
                       const std::string    &projectname = "");
     
+    PrintObjects            objects() { return m_objects; }
 private:
     
     // Implement same logic as in SLAPrintObject

```
It's forcefully saving the mesh+support as `/tmp/xxx.obj` successfully. 
I'll try to convert this to a `--export-stl-with-supports` option and pull request it.

k-gordon, 2024-03-21:
> I was able to make it work in a very "hack" way today: [cli export obj with support](https://github.com/hradec/PrusaSlicer/commit/c8c90968f5f1779b08ecece43586139d35df2ea0)

That's awesome @hradec!

closed 2026-09-29 by prusaslicer-bot[bot]