prusa3d/PrusaSlicer #5685: Supports instead of infill / infill as support

lukasradek, 2021-01-04:
[ Feature Request ]

### Use case
Print object as a hollow shell (with open bottom) from a solid model. As opposed to vase mode, this would allow for multiple perimeters and overhangs.

### Principle
Create similar infill to shell interface as with support to part interface, so that infill can be **easily removed**. It would be great to combine this with Adaptive Support Cubic, so that support infill can be sparse "at the bottom" and dense near the supported surface for better quality.

### Result
Open bottom shell with easily removable internal infill / supports, which would get denser with proximity to supported layers.

Huge thanks for the continuous development 😉.

bubnikv, 2021-02-24:
Frankly I don't understand. A picture would help.

lukasradek, 2021-02-24:
I am not sure, image will help much

Minimal example:
Imagine half a sphere solid object laying on it's flat side on the bed in slicer.

Normally it would print as solid hemisphere consisting of hemispherical shell, a flat base and some inner infill.

I want the result to be the hemispherical shell only.

I can prevent the flat base from printing by reducing bottom layers to 0.
I need to print the infill, otherwise structure would not hold, but removing the infill requires force.

Therefore if I can print infill that interfaces with the shell the same way supports do (z distance), I will be able to remove it much easier.

Simply put - allow support algorithm to be used as infill pattern inside solid objects. Further combination with adaptive cubic would make it even better (efficient).

Hope I explained it better.

bubnikv, 2021-02-24:
Why don't you just hollow your object in your CAD when designing it?

lukasradek, 2021-03-18:
I am sorry for that delay.

The reason is quite simple. Especially relevant when thin walls are required.

### Hollowing in CAD
1. Hollowing a complex shape can be tricky (corners, keeping thickness of the wall).
2. Resulting wall thickness may not translate nicely into perimeter count in slicer. Since the slicer has to fit nozzle passes between the inner and outer wall, it sometimes results in unevenly filled space between perimeters or missing perimeters, where the thickness becomes to low.
### Using infill as support
1. Simply create solid object
2. In slicer, infill is subordinated to the perimeters, so slicer has to consider only the outer wall, when planning nozzle passes. Therefore perimeter count (and wall thickness) will be uniform and the whole process is much simpler.

And if the implementation might be just (an educated guess, might be wrong)

a) allowing the infill to be generated by support algo OR
b) allowing the already implemented supportive cubic to have a "Contact Z distance"

, it is quite a low hanging fruit IMHO.

closed 2026-09-29 by prusaslicer-bot[bot]