prusa3d/PrusaSlicer #5687: Weird think lines

Thomas-Sky, 2021-01-04:
PrusaSlicer 2.3.0-rc1+win64 Windows 10

Printing cone vase got weird thick random lines along perimeter
The issue appears any walls perimeter applied.
[image]
[image]
On the other side instead everything is ok:
[image]

Here goes the Prusaslicer settings:
[image]))
[image]
[image]

The printer and slicer work perfect on other type of objects, I get quality really good, even with filament used for this cone model.
Attached the 3mf file for discovering more

rtyr, 2021-01-05:
Please provide 3mf project file (it is not attached).

Thomas-Sky, 2021-01-05:
Sorry, my mistake

[3_1_w1rf_Shell-CUT.zip](https://github.com/prusa3d/PrusaSlicer/files/5769713/3_1_w1rf_Shell-CUT.zip)

rtyr, 2021-03-01:
The wall thickness is a little bit small for your extrusion width. I would try it with reduced extrusion width for perimeters/external perimeters.

closed 2026-09-29 by prusaslicer-bot[bot]