prusa3d/PrusaSlicer #5688: The Octoprint connection seems to be missing

srjacob999, 2021-01-04:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

srjacob999, 2021-01-04:
Using 2.3.0 rc2 with a Prusa i3MK3/S.
The connection to Octoprint seems to be missing.

lukasmatena, 2021-01-05:
From [2.3.0-alpha1 change log](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-alpha1):

> Starting with this version, PrusaSlicer separates the physical printer connection settings from the Printer profile into new Physical Printer profiles. A Physical Printer profile references one or more Printer profiles of the same technology (either FFF or SLA), and it assigns a network connection information to these printers. One does not have to create a Physical Printer profile if one does not need to connect to a printer over the network. The new Physical Printers are listed in a new section in the Printer selection combo box after the system and custom Printer profiles.

> A new Physical Printer profile is created by clicking on the "cog" icon right to the Printer profiles combo box, or by selecting the "add or remove printers" item in the Printer combo box. The Physical Printer profile editor opens also when clicking on the new "cog" icon in the Printer settings tab. The Physical Printer profiles are being stored into PrusaSlicer/physical_printer directory.

> When first installing PrusaSlicer 2.3.0-alpha1, the network connection information is removed from existing user Printer profiles and converted to new Physical Printers profiles giving them generic names "Printer 1, Printer 2, ...". The user is advised to review and rename these Physical Printer profiles generated by the conversion process.

Does it solve the problem?

srjacob999, 2021-01-05:
No.  I already tried that and the Octoprint connection still isn't there.

lukasmatena, 2021-01-05:
Could you please be more clear about the "still isn't there"? What is not present and where? Were you not able to add a physical printer profile? Did you try to use your old config, but it was not converted correctly?

srjacob999, 2021-01-05:
OK, I found it and created another printer (the information from my Prusaslicer 2.2) was not copied over.  Problems so far.

1)      Octoprint find couldn’t find my Octoprint

2)      When I entered the IP address of API Key for my Octoprint, and hit test, it seemed to find it ok.  Thankfully, I use a static IP address for Octoprint.  If I had used DHCP, I would have had to find Octoprint another way.

It would have been nice to have Prusaslicer pick up my Octoprint settings and import them.

 

I’ll start using the new Prusaslicer and see how it works.  I wish the support material section had been improved (like Simplify 3D).

 

Also, none of my own presets (for print settings, filament settings, etc.) were migrated over to the new version.  This is unfortunate because I had the support material parameters for PETG nailed down in V2.2, and now they are gone in V2.3.

 

I think perhaps I’ll stay with V2.2 for a while.

 

Steve

bubnikv, 2021-09-22:
> 1)      Octoprint find couldn’t find my Octoprint

Did it ever work? Does your router support Bonjour discovery?

> It would have been nice to have Prusaslicer pick up my Octoprint settings and import them.

That should have worked. Anyways, I think most of the users already converted to 2.3, so it is no more an issue I suppose.

bubnikv, 2021-11-08:
No response, closing.

closed 2021-11-08 by bubnikv