prusa3d/PrusaSlicer #5693: Add Ender3V2 and Das Filament PLA & PETG to Creality.ini

etofi, 2021-01-05:
I have included the Ender 3V2 in the config and the corresponding PLA and PETG from "Das Filament". Can you please include this?

[ENDER3V2.zip](https://github.com/prusa3d/PrusaSlicer/files/5771923/ENDER3V2.zip)

closed 2021-01-07 by etofi

pmjdebruijn, 2021-01-23:
How did you get to 215C for Das Filament PLA?
Did you print a temperature tower? ( https://www.thingiverse.com/thing:2615842 )

etofi, 2021-01-23:
> 
> 
> How did you get to 215C for Das Filament PLA?
> Did you print a temperature tower? ( https://www.thingiverse.com/thing:2615842 )

I printed [this](https://www.thingiverse.com/thing:2729076) one.

pmjdebruijn, 2021-01-23:
Fair enough. Thanks for clarifying!

pmjdebruijn, 2021-01-24:
Oh I almost forgot, hopefully did you perform the test at 0.1mm? 

Issues with regard to retractions and bridging tend to be masked by larger layer heights.

etofi, 2021-01-25:
No, I printed it in normal quality with 0.2
But it is generally better to print a head tower yourself. Everyone has different environmental conditions at the printer location.

pmjdebruijn, 2021-01-25:
I'm not asking for my own use. All the other filaments I've added were done at 0.1, as 0.2 can hide issues related to bridging/retractions. There's a fair chance, the presets should be 5C lower. Would you mind doing the temperature towers again at 0.1mm?

etofi, 2021-01-28:
Today I printed a new PLA Heattower with 0.1.
Here is the link to the [pictures](https://cloud.troeschs.de/index.php/s/fXLq4ZmzW6Cf8wX). What do you think which temperature is the best?

pmjdebruijn, 2021-02-01:
@etofi thanks for sharing the pictures. And the results are remarkably close. Presumably the improved part cooling fan duct on the V2 plays a part in that.

Given, that these filament presets are shared between all Creality printers, including those with older part cooling fan duct designs, I would suggest lowering the regular print temp by 5C:

temperature = 210
first_layer_temperature = 215

@rtyr: if @etofi doesn't object, would you mind adjusting Das Filament PLA?

rtyr, 2021-02-01:
ok, done

pmjdebruijn, 2021-02-15:
@etofi, what are the physical dimensions (so the actual measurements) of the print bed on the Ender-3 V2?

etofi, 2021-02-15:
[See offical Website](https://www.creality3dofficial.com/de/products/ender-3-v2-3d-printer)...
220x220x250

rtyr, 2021-02-15:
https://overclock3d.net/news/misc_hardware/creality_s_ender_3_v2_3d_printer_upgrades_an_industry_icon/1

> Like older models of the Ender 3, the V2 offers users a 220mm x 220mm x 250mm (X x Y x Z) build area, slightly undercutting the printer's maximum bed size of 235mm x 235mm. Like the original Ender 3, the V2 can likely use this extra area using custom firmware, though this alteration is not recommended for beginners.

pmjdebruijn, 2021-02-15:
@etofi: I'm asking about the measured dimensions, not Creality's marketing. As far as I can tell the firmware is indeed, sadly, limited to 220x220, however, the physical bed is likely larger, as @rtyr suggests it's probably 235x235, however, recently creality has been making some beds (particularly the glass ones) slightly deeper than wide, because of the clips.

Would you mind actually taking a measuring tape to your print bed, and reporting the actual physical dimensions?

etofi, 2021-02-17:
235x235 is correct