prusa3d/PrusaSlicer #5694: Multi Color model doesn't address all items correctly

ErikV1000, 2021-01-05:
### PrusaSlicer-2.2.0+win64-202003211145
### Operating system type + version: Win 10 64
### 3D printer brand / version + firmware version (if known): Prusa MK3S/MMU2S version 3.9.2
### Behavior
I have imported a Multi STL cell model from Thingiverse (thing:2485063) and set it up in Prusa slicer by using the Cell_Outer.stl as base and added all the other items to the base, as shown in the picture. Then I linked the different parts to different colors in the MMU and slided the model. The slice proces however doesn't attach one of the stl's to the assigned color, and leaves it white (color 1). I have tried to add the parts in a different order, but still the same part is wrongly colored. When slicing the part on its own there are no issues and it will accept any color from the MMU.

The file printed fine and resulted in a nice cell model, but not completely as planned.

I do wonder what went wrong and how this can be corrected,

Looking forward to hear from you.

[image]

[image]

[Prusa System Information dialog.txt](https://github.com/prusa3d/PrusaSlicer/files/5771981/Prusa.System.Information.dialog.txt)

#### Project File (.3MF) where problem occurs
[cell_complete.zip](https://github.com/prusa3d/PrusaSlicer/files/5771999/cell_complete.zip)
[cell_complete_0.15mm_PLA_MK3SMMU2S_1d5h28m.zip](https://github.com/prusa3d/PrusaSlicer/files/5772003/cell_complete_0.15mm_PLA_MK3SMMU2S_1d5h28m.zip)

bubnikv, 2021-01-06:
I think the missing red lines are just too thin to extrude. You may try to
scale up the model, or to enable "detect thin walls"

ErikV1000, 2021-01-06:
Hi Vojtěch,

Thanks for that suggestion. I have sliced it again and this seems to work.

Regards, Erik
[image]

bubnikv, 2021-01-07:
thanks, closing.

closed 2021-01-07 by bubnikv