prusa3d/PrusaSlicer #5695: Slicing large object on MacOS V10.11 doe not work any more with high infill.

chmuc01, 2021-01-05:
### Version
2.3.0-rc2+

### Operating system type + version
Mac OS X Version 10.11.6 (El Capitan)
Resident memory: 3,406MB
Peak memory usage: 4,725MB
RAM size reserved for the Undo / Redo stack: 859MB
RAM size occupied by the Undo / Redo stack: 0MB
OpenGL installation
GL version: 2.1 NVIDIA-10.0.52 310.90.10.05b46
Vendor: NVIDIA Corporation
Renderer: NVIDIA GeForce 9400M OpenGL Engine
GLSL version: 1.20
Installed extensions:
GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_fragment_program2
GL_NV_fragment_program_option
GL_NV_light_max_exponent
GL_NV_multisample_filter_hint
GL_NV_point_sprite
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
Eigen vectorization supported: SSE, SSE2, SSE3, SSSE3

### 3D printer brand / version + firmware version (if known)
Original Prusa I3 MK3S FW V3.9.1

### Behavior
* Problem: Slicing a large object (150mm x 80 mm x 140mm with a large Infill (e.g. 70%) created in my environment the following problems: After the slicing has finished, the legend (with the times) is not displayed within the 3D display and the whole MacBook screen clickers on every move of the cursor within the 3D display area. The display area does not react any more (no change of the size or view, e.g. zoom, rotate, pan). Changing back and forth to the Platter and slicing result display lets the PrusaSlicer crash. Sometimes at this point also the OS freezes fully so that the MacBook has to reset hard.
* Load a large object and slice it with 70% infill (Gyroid).
* Slicing should finish without the problems, the slicing should be finished with the display of the legend within the 3D display area and the 3D area should be responsive.
* Slicing is finished without the last display update and 3D view is frozen and is missing the legend.

_Is this a new feature request?_ NO

#### Project File (.3MF) where problem occurs

chmuc01, 2021-01-05:
I'm don't know how to upload the 3mf file here ...

bubnikv, 2021-01-06:
zip it or just add a .zip extension and drag & drop it here.

chmuc01, 2021-01-09:
Just tried the same project file with RC3 on a MacMini (M1) under BigSur 11.1.
The project loads and slices but showing the final sliced 3D preview takes a while (> 60 sec after slicing is finished).
Unfortunately the 3D preview seems frozen. Moving the mouse cursor within the 3D area changes to mouse cursor to the spinning wheel for approx. 7 sec. So I can not change the view (e.g. rotating, zooming) with the mouse. Switching back to the 3D editor view works after clicking on the icon at the bottom left 2 times and waiting a while. This view is working and changing the view (rotating, zooming) works fine. Switching back to the preview shows the view but it's still not responsive to mouse. Using shortcuts (e.g. 0, 1, 2, 3) changes the view after a spinning cursor and about 10 seconds.

chmuc01, 2021-01-09:
Just recognised that it seems that the gcode viewer is the problem and it is totally 'overloaded' with this object sliced with 70% infill (Gyroid), resulting in a 233MB gcode file. At least on my two MAc machines.
On my Windows PC (Windows 10, AMD Threadripper 1920X, GTX 1080 Ti) the read of the gcode file takes longer than on the two Macs but the 3D view of the final result (70% infil, gyroid) works like a charm.

Can it be that you still have some issues with the 3D preview on Mac?

Finally: It seems that m project works without any problems on WIndows 10 and a NVIDIA 3D card but does not support 3D cards on (current) Mac machines...

bubnikv, 2021-01-11:
> Can it be that you still have some issues with the 3D preview on Mac?

See https://github.com/prusa3d/PrusaSlicer/issues/5412

We are aware of the issue, we are working on a fix, however Apple obsoleted OpenGL, thus they do not really help us resolving the issue. We feel like alchemists, thank you Apple.

chmuc01, 2021-01-13:
@bubnikv you are right. Seems to be the identical issue.
For Apple devices a move to 'Metal' seems to be the only real solution as OpenGL was already EOL in 2018.
So I accept that we can close the issue as a duplicate of #5412.

closed 2021-01-13 by chmuc01

bubnikv, 2021-04-14:
Fixed in PrusaSlicer 2.3.1-rc