prusa3d/PrusaSlicer #5696: \Appdata\Roaming\PrusaSlicer\PrusaSlicer.ini corrupted T

dman6186, 2021-01-06:
Hello same corrupt file, but nothing to do with WIFI like the post below. 

I tried to uninstall and reinstall by still the same pop-up appears.

Any ideas to help?
Thanks in advance.

Thank you for the report, I'll fix this...

_Originally posted by @vojtechkral in https://github.com/prusa3d/PrusaSlicer/issues/2927#issuecomment-531820680_

bubnikv, 2021-01-06:
You have to delete the file that the error message refers to, PrusaSlicer.ini
Also please post the PrusaSlicer.ini here before deleting, we are interested in its content. This is something we were not able to resolve for a long time. It happens extremely rarely, but it happens.

5jvm0u4, 2021-01-08:
I have similar problem that shows the missing of Creality vendor .ini file when I open PrusaSlicer 2.2.0 and 2.3.0 RC2 simultaneously and it told me that I need to reconfigure due to incompatibility of my profiles. Here are some snapshot of the error message that showed up. I've checked that folder and it's empty without me delete any file.
[image]
[image]

bubnikv, 2021-01-08:
Would you please zip and share the content of your PrusaSlicer directory
with us, so we can analyze the data issue?

5jvm0u4, 2021-01-08:
I've uninstalled my 2.2.0 and deleted everything, so I only got this from my recycle bin, hope this would help.
[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/5787962/PrusaSlicer.zip)
It's the folder in User>AppData>Roaming that I deleted 3 hours ago.

bubnikv, 2021-01-08:
PrusaSlicer 2.3.0-rc3 starts with the attached PrusaSlicer configuration
directory just fine.

bubnikv, 2021-06-25:
This issue is stalled. Do you guys still have the issue? Could you share data for us to test with?

turgaycinbat, 2021-11-16:
> Merhaba aynı bozuk dosya, ancak aşağıdaki gönderi gibi WIFI ile ilgisi yok.
> 
> Kaldırmaya ve yeniden yüklemeye çalıştım, hala aynı açılır pencere beliriyor.
> 
> Yardımcı olacak herhangi bir fikir var mı? Şimdiden teşekkürler.
> 
> Rapor için teşekkürler, bunu düzelteceğim...
> 
> _İlk olarak @vojtechkral tarafından [# ](https://github.com/prusa3d/PrusaSlicer/issues/2927#issuecomment-531820680)2927'de gönderildi [(yorum)](https://github.com/prusa3d/PrusaSlicer/issues/2927#issuecomment-531820680)_

[General]
Flags=PerMachine|ShowConfigOptionsButton|InstallSummaryAlways
AppDir=C:\Users\
ApplicationName=PrusaSlicer 2.3.3
CompanyName=Prusa Research
ApplicationVersion=2.3.3
DefaultCommandLine=/checknow
CheckFrequency=2
DownloadsFolder=C:\Users\Public\Documents\Prusa3Dupdater\updates\
ID={936E2B15-E838-4294-B70B-C3E03A66ACDE}
URL=https://www.prusa3d.com/downloads/update.txt?test

turgaycinbat, 2021-11-16:
> Merhaba aynı bozuk dosya, ancak aşağıdaki gönderi gibi WIFI ile ilgisi yok.
>
> Kaldırmaya ve yeniden yüklemeye çalıştım, hala aynı açılır pencere beliriyor.
>
> Yardımcı olacak herhangi bir fikir var mı? [...]

I set a date back to 1 week without deleting the slicing program, I created an installation point, after I restored it, I gave a prusaslicer 2.3.3 command again and it was fixed.

turgaycinbat, 2021-11-16:
I set a date back to 1 week without deleting the slicing program, I created an installation point, after I restored it, I gave a prusaslicer 2.3.3 command again and it was fixed.

turgaycinbat, 2021-11-16:
I set the date back 1 week without deleting the slicing program, I created an installation point, after I restored it, I gave the prusaslicer 2.3.3 installation command again and it was fixed.

turgaycinbat, 2021-11-16:
Without deleting the slicing program, I set my pc back 1 week, restored it, created an installation point, after I restored it, I gave the prusaslicer 2.3.3 installation command again and it was fixed.

turgaycinbat, 2021-11-16:
> Merhaba aynı bozuk dosya, ancak aşağıdaki gönderi gibi WIFI ile ilgisi yok.
>
> Kaldırmaya ve yeniden yüklemeye çalıştım, hala aynı açılır pencere beliriyor.
>
> Yardımcı olacak herhangi bir fikir var mı? [...]

Without deleting the slicing program, I set my pc back 1 week, restored it, created an installation point, after I restored it, I gave the prusaslicer 2.3.3 installation command again and it was fixed.

closed 2026-09-29 by prusaslicer-bot[bot]