prusa3d/PrusaSlicer #5698: When editing presets, list items are always green even if values have changed compared to preset or defaults

lindequist, 2021-01-06:
### Version
2.3.0-rc2+win64

### Operating system type + version
Windows 10, 64 bit

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S with MMU2S (probably not relevant)

### Behavior
* In 2.2.0 the text of the list items to the left of the screen when editing settings would be green if all values on that "tab" (e.g. Layers and perimeters or Filament overrides) were identical to the preset values, and the list item text would be black if any values were changed. This feature made it easy to identify changed settings for presets being derived from other presets, and it was very useful.
* In order to reproduce the problem, just edit e.g. an existing filament preset and save it under a different name. The list item  containing the changes should be indicated with black as it did in 2.2.0 but in 2.3.0-rc2 the list item text is green and thus doesn't highlight the changed value. The locks and bullets right next to the value appear to be working fine.

#### Project File (.3MF) where problem occurs
[image]

bubnikv, 2021-01-07:
The value is first accepted when you leave the field.

lindequist, 2021-01-07:
> […](#)

I know, and that's why I wrote that the preset should be saved after changing the value. After saving the preset, the "Advanced" list item would in 2.2.0 become black due to the value deviating from the original preset, but in 2.3.0-rc2 that "Advanced" list item stays green. This new behaviour makes it harder to spot modified values in presets derived from other presets via a dependency.

bubnikv, 2021-01-08:
Fixed in PrusaSlicer 2.3.0-rc3.

closed 2021-01-08 by bubnikv