prusa3d/PrusaSlicer #5699: Jams when printing at 0.05mm

markdesouza, 2021-01-06:
### Version
PrusaSlicer 2.3.0-rc2+

### Operating system type + version
Mac OS 11.1

### 3D printer brand / version + firmware version (if known)
Printer: Prusa MK3s+
Firmware: 3.9.2
Nozzle: 0.4

### Behavior
On PrusaSlicer 2.2 I was able to print at 0.05 without issue (although it was different models)

After upgrading to PrusaSlicer 2.3.0-rc2, when I print at 0.05mm (default setttings, with detect thin walls enabled) I get a filament removal request from the printer during the end of the first layer. Presumably due to a thinking there is an issue with the filament/print ie clog. I have tried reslicing & printing the file 3 times and the same issue occurs. 

Reslicing at 0.1mm (default setttings, with detect thin walls enabled) the print successfully prints first time. Same filament, same nozzle, same model

#### Project File (.3MF) where problem occurs
Unfortunately I can not submit the file as its purpose is commercial in nature.

rtyr, 2021-01-07:
I don't think your issue is caused by the PrusaSlicer. There is no clog detection on MK3, only filament runout detection. You can try to print it with disabled filament sensor (can be disabled from LCD menu).
Other possibility is accidentally added color change in the PrusaSlicer (M600). I am afraid that's all what I can say without any data. Since it happens after first layer, maybe you can cut the model and provide the gcode/3MF.

markdesouza, 2021-01-07:
@rtyr I agree, I thought it improbable but something is definitely happening. Given its only in 2.3 in the first few layers maybe its related to the monotonic infill?

> There is no clog detection on MK3, only filament runout detection
Hmm if it happened once, I could understand, but I had 3 failed prints in a row, when printing at 0.05mm and now 4 successful prints printing at 0.1mm. So the filament must be fine and in all cases the filament hadn't run out.

> Other possibility is accidentally added color change in the PrusaSlicer (M600)
I just grepped both gcode files and they both only contain on line with M600 and thats the usual commented out line
; color_change_gcode = M600
So no color changes....

rtyr, 2021-01-07:
If you are asked to change the filament and there is no M600 in the gcode then the filament change is trigerred by filament sensor. You can simply test it by disabling it.

rtyr, 2021-01-14:
Closing. Likely not PrusaSlicer related issue.

closed 2021-01-14 by rtyr