prusa3d/PrusaSlicer #5700: When printing with a Brim + Supports, the brim collides with the purge border

scottmudge, 2021-01-07:
### Version
2.3.0-rc2

### Operating system type + version
Windows

### 3D printer brand / version + firmware version (if known)
MK3S+

### Behavior
When support enforcers are used to produce supports for an object, and when "Brim" is enabled, the brim collides with the initial purge perimeter of the object, and is overprinted on top of it:

[image]

If Brim is turned off, the perimeter can be seen not collided or overprinted:

[image]

Perhaps there is a method of preventing this behavior, but it appears automatic.

neophyl, 2021-01-07:
This has been reported numerous times.  Its intentional, by design apparently and in response to user request.

scottmudge, 2021-01-07:
What's the intention? Shouldn't there at least be a warning that the material will be overprinted and may cause issues with the print? Perhaps it's not actually overprinting, but instead the brim is simply being broken up? It's difficult to tell.

At the very least there should be an option to perform an offset operation for the purge perimeter/border. Or if the brim makes the purge perimeter redundant (given it serves a similar function), just disable it entirely

neophyl, 2021-01-07:
You would have to address that question to a Prusa Dev.
Some of the previous issues 
#3962
#3516
#4017
#4024
#4185
#4056
#4354
#4506

bubnikv, 2021-09-22:
The behavior was changed in PrusaSlicer 2.4.0. If the skirt is NOT a draft shield, it is printed now outside the brim.

closed 2021-09-22 by bubnikv