prusa3d/PrusaSlicer #5701: Print Profiles not available

maot1985, 2021-01-07:
### Version
2.2.0+64

### Operating system type + version
Windows 10 build 18363 64-bit

### 3D printer brand / version + firmware version (if known)
Renkforce RF2000v2

### Behavior
Today I opened PrusaSlicer and my printing settings are not available. The files are still in the print-folder, but they do not appear in PrusaSlicer. The filament and printer settings are available. I went to my Laptop and the same error occurs!

rtyr, 2021-01-07:
We can look at it if you provide your archived data folder.

maot1985, 2021-01-07:
I zipped the folder C:\Users\User\AppData\Roaming\PrusaSlicer\
The file DKM(%).ini does not appear in PrusaSlicer anymore.

[https://github.com/maot1985/PrusaSlicer-Problem](url)

rtyr, 2021-01-07:
The DKM(%) print profile is visible only for RENKFORCE_SINGLE2 (because only this printer meets the condition set for DKM print profile). You have to modify the condition (or printer_notes section in RENKFORCE_SINGLE1 printer profile).

[image]

maot1985, 2021-01-07:
Ok, thank you very much :-) The strange thing is that the profile worked for both printers in the past ...

rtyr, 2021-01-07:
You probably changed printer_notes at some point.

rtyr, 2021-01-07:
I believe we can close this.

closed 2021-01-07 by rtyr

maot1985, 2022-01-01:
He KatzeMau, I do not use the dual print option on the RF2000v2. I tried it once with the Repetierhost profiles from Conrad, but for me it is too much work and time consuming. With the prusaslicer it did not work for me.