prusa3d/PrusaSlicer #5703: Ironing all solid surface does not seem to work

Thelvaen, 2021-01-07:
### Version
2.3.0-rc2+GTK3

### Operating system type + version
Linux

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
* Ironing all solid surface does not seem to work
* I tried to print a translucid Benchy, using ironing on all solid surface, with 100% infill, but as far as preview is concerned, 2h40, is a bit to short to get all the 320 layers printed & ironed.

#### Project File (.3MF) where problem occurs
see attached Benchy project

lukasmatena, 2021-01-07:
> see attached Benchy project

I don't see any attached project.

Thelvaen, 2021-01-07:
Sorry I had some trouble attaching the .3mf file. It should be fixed now.

wabbitguy, 2021-01-08:
I filled a block with 100% infill and enabled ironing on all solid surfaces and it only ironed the top surface. Mac OS X 10.15, RC2

Mel

MDReptile, 2021-01-12:
I also cannot get every layer ironing to work, only on top layers. Win 10, latest stable release of prusaslicer.

https://help.prusa3d.com/en/article/ironing_177488
^ this page says that "This option is not working properly at the moment."

bubnikv, 2021-01-12:
Ironing all layers does not work yet. We will fix it in the upcoming minor release.

bubnikv, 2021-08-06:
Implemented in upcoming PrusaSlicer 2.4.0-alpha1.
Closing.

closed 2021-08-06 by bubnikv