prusa3d/PrusaSlicer #5704: Still issues with support and option to turn of bridging

TheDeadPrinter, 2021-01-07:
### Version
Version 2.3.0-rc2+win64

### Operating system type + version
_Windows 10 64bit
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
The support under the printed parts is not under the whole overhang, resulting in ugly, loose overhangs. There is a huge gap between the support zigzags (in the middle).
[image]

Bridging layer is way to high as you can see in the picture and cannot be turned off at all, I already tried adjusting the flow und uncheck "Detect bridging perimeters" Bridging is still there (light blue). 
[image]

[image]

The support interface doesn't help (I already tried turnig it 45° or 90°) The gap between the printed part is too big (my setting is 0,1mm, but it doesn't seem to be that way)
[image]

Looking at Cura 4.8 (and many versions before that) The support interface looks better (can be changed) and the support overhangs look way better too. But I'd really like to use PrusaSlicer for my Prusa printer (because I prefere the overall print and compatibility with my i3 mk3s)
[image]
The lower one is Cura (with ste smaller support structure)
[image]
[image]
[image]

I hope these issues will be resolved in the releasing PrusaSlicer 2.3

Greetings
TheDeadPrinter

rtyr, 2021-01-07:
> The support under the printed parts is not under the whole overhang, resulting in ugly, loose overhangs. There is a huge gap between the support zigzags (in the middle).

Disable "Don't support bridges".

rtyr, 2021-01-08:
I believe the reason for the gap was explained. Closing.

closed 2021-01-08 by rtyr

TheDeadPrinter, 2021-01-09:
Oh, I forgot to mention that setting. Even with that setting disabled the overhang looks the same loose way. I also tried setting the "contact Z distance" to 0.05 and 0.01, which improved the overhang a bit, compared to the 0.1 Z gap, With the Z gap set to 0 (which is for soluble filament) and synchronized layer hight, the overhang bonds with the overhang nearly inseparable.
I think that there should really be a way to turn off bridge infill.