prusa3d/PrusaSlicer #5706: [Enhancement Request] Cut multiple Dimensions Easier

colsowka, 2021-01-07:
### Version
2.3.0 Beta 2

### Operating system type + version
Windows 10 64 bit
### 3D printer brand / version + firmware version (if known)
MK3S+

Currently the process for cutting in the other dimensions other than Z is to

1. Cut Z
2. Place on Y face
3. Cut Y
4. Place on X face
5. Cut X
6. Place on face

I propose we simplify this significantly by having an optional tick box for the cut tool that allows two other gizmos for the X and Y planes.

[image]

jmsaltzman, 2021-11-09:
This would be a great addition, and there's already an options GUI panel for the tool.

My use case: a portion of a large model is not slicing well, so the section is separated for test printing while fine tuning parameters. @neophyl Since it is not an issue with the model, but with the slicing, I believe this belongs in the slicer to reduce cycle time and steps needed. I am familiar with the steps listed by @colsowka, often taking them multiple times in a short period.

lukasmatena, 2023-02-01:
Implemented in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena