prusa3d/PrusaSlicer #5708: M73 Percentage not gradual on small prints

WalnutFox, 2021-01-08:
### Version
Version 2.3.0-rc2+win64

### Operating system type + version
Windows 10 (Build 19042)

### 3D printer brand / version + firmware version (if known)
Ender 3 Custom Firmware

### Behavior
M73 Percentage jumps big time on small prints (<10 minutes)

_Steps to Reproduce_
1. slice anything less than 10 Minutes
2. inspect m73 in gcode

_Expected Result_
Gradual increase of percentage

_Actual Result_
Jumps from a number like 69% to 100% even though gradual before hand.

#### Project File (.3MF) where problem occurs
[test.zip](https://github.com/prusa3d/PrusaSlicer/files/5785423/test.zip)

WalnutFox, 2021-01-08:
Here are the last few m73 commands sent by the print so that you can better understand what i meant:
```
M73 P65 R3
M73 P66 R3
M73 P67 R3
M73 P68 R3
M73 P69 R3
M73 P100 R0
```

enricoturri1966, 2021-01-08:
@WalnutFox

The problem seems to be caused by the line:
G1 Y2900 F1000;
in your custom End G-code.

The estimated time for this line is: 167 seconds, which is more than 30% of the total estimated time.
This is why there is such a jump in the percent reported in the M73 lines.

[image]

WalnutFox, 2021-01-08:
Strange I would have thought that the time calculation would ignore the start and end gcodes. Maybe it is just me but that would sound more like a proper calculation.

_suggested fix_
M73 calculation should apply to sliced object only, ignoring start and end gcodes.

lukasmatena, 2021-01-08:
> Maybe it is just me

Maybe. I would expect that they are taken into account.

My suggested fix: remove a line that takes three minutes to finish and which is probably just ignored by your firmware because it is unable to process it anyway (given actual dimensions of Ender-3 print area). What is its purpose?

WalnutFox, 2021-01-08:
I can just move the line to octoprint end code, it just moved the build plat forward. I never bothered to put the right dimension. This is technically my fault but I still don't understand why the m73 calculation would need to take into account the end and start code.

neophyl, 2021-01-08:
because printing isnt actually finished until ALL code has been processed I would assume ?  Why shouldn't it take ALL code into account ?

WalnutFox, 2021-01-08:
So my partner just explained to me a use case that makes sense for all code to be taken into account. If your running a print farm you want p100 as the last thing so you know when the printer is ready for use again.

lukasmatena, 2021-01-11:
Ok, I'm closing the issue, it's working as expected.

closed 2021-01-11 by lukasmatena