prusa3d/PrusaSlicer #5709: PrusaSlicer 2.3.0-rc2 issue with skirt & brim overlapping

mfontaine3277, 2021-01-08:
### Version
PrusaSlicer 2.3.0-rc2

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Prusa MK3S+

### Behavior
I had a model with a very small footprint and some supports.  I added a large brim (5mm) to give the object greater bed adhesion.  When printing, I noticed the the brim printed over and beyond the skirt.  This is also visible in preview mode, I just didn't notice it until printing.

[image]

foreachthing, 2021-01-08:
rtyr  says in https://github.com/prusa3d/PrusaSlicer/issues/5383:
> **Intended behavior.**
> 
> Why:
> #724
> 
> Duplicate issue + links to many other similar issues:
> #3692

rtyr, 2021-01-08:
Yes . See the mentioned issues for more information or further discussion. Closing.

closed 2021-01-08 by rtyr