prusa3d/PrusaSlicer #5710: Perimeter and infill speed changes with certain types of infill despite the numeric values not changing

vguttmann, 2021-01-08:
### Version
_2.2.0_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Ubuntu 20.04 LTS_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_Ender 3, no HW modifications, Marlin 2.0.7.2 w/ custom config_

### Behavior
* _Describe the problem:_ Perimeter speeds should be as set in the Print settings, but changes over the Z axis.
* _Steps needed to reproduce the problem:_ Use any type of infill that changes over Z (Gyroid, 3D Honeycomb, Cubic), then view the speed of the sliced print.
* _Expected Results:_ The infill and perimeter speeds are always the same, and are the numbers set in the print settings
* _Actual Results:_ Both the infill and the perimeter speed vary over the course of Z
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
 [image]
[image]

_Is this a new feature request?_ No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

[cube.zip](https://github.com/prusa3d/PrusaSlicer/files/5787015/cube.zip)

_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2021-01-08:
because your object is small, thus the cooling slow down kicks in. How much
it slows down depends on the particular shape of the infill at a particular
layer.

rtyr, 2021-01-08:
https://help.prusa3d.com/en/article/cooling_127569

vguttmann, 2021-01-08:
But I set the perimeter speed, and theoretically it should print at that speed. Is there a setting to make it print at the usual speed and then just retract, move a bit up and wait? Slower speeds give me all sorts of problems, from bridging to a bad surface finish, so I'd prefer if it would not slow down so much, and instead wait afterwards.

vguttmann, 2021-01-08:
Oh, thanks!

closed 2021-01-08 by vguttmann