prusa3d/PrusaSlicer #5712: G-code Viewer / Mac / COMMAND+M

dahuafschmied, 2021-01-08:
### Version
2.3.0 RC3

### Operating system type + version
Mac OS

### 3D printer brand / version + firmware version (if known)
-- wurscht

### Behavior
In G-Code Viewer Command+M is for Minimize the window. But it should open the 3DConnexion Dialog if enabled in settings.

[image]

#### Project File (.3MF) where problem occurs
-- a wurscht

dahuafschmied, 2021-09-01:
Command+m gone in 2.4

closed 2021-09-01 by dahuafschmied

bubnikv, 2021-09-02:
It is in the change log

macOS specific: Enabled the platform standard "Cmd+M" hot key to minimize the application. The "3DConnexion legacy set sensitivity dialog" hot key was changed to "Cmd+Shift+M" on macOS.