prusa3d/PrusaSlicer #5714: SLA Optimize Orientation Not Working as Expected

nolspeed29, 2021-01-08:
### Version
2.3.0-rc2+

### Operating system type + version
macOS Catalina 10.15.7

### 3D printer brand / version + firmware version (if known)
Prusa SL1

### Behavior
* _Describe the problem_
Since updating to PrusaSlicer 2.3.0, the Right Click>Optimize Orientation no longer works as it did previously.  The model is now just set orthogonal on the build platform.  It does not rotate to an optimism (usually tilted back) orientation like before.  
* _Steps needed to reproduce the problem_*
Import .stl model, Right-Click>Optimize Orientation. Using default 0.05 Normal profile and material.
* _Expected Results_
Model gets orientated in a complex manner to minimize supports and enhance print reliability
* _Actual Results_
Model gets positioned orthogonally and does not change with repeated attempts.
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
  

[image]
_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs

[CatfolkRogue.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5787824/CatfolkRogue.3mf.zip)

tamasmeszaros, 2021-01-08:
Try to enable "pad around object" mode on the side panel and optimize orientation after that.

nolspeed29, 2021-01-08:
I did this and now the part reoriented w/ the base flat.  Continued attempts do not change the orientation.

[image]

Area5142, 2021-01-09:
I can confirm the problem with _"Optimize orientation"_.

Here the model orientation is optimized in PS 2.2.0:
[image]

And here the model orientation is optimized in PS 2.3.0RC3:
[image]

Configurations for the models are the same.

nolspeed29, 2021-01-09:
Just confirmed this issue on my Windows64 PC as well.  Fresh install using default profile and both pad settings.

tamasmeszaros, 2021-01-11:
The optimize rotation feature was changed in this release (by me). The old behavior was optimizing for the greatest misalignment with all the axes so that the print would (maybe) end up having smoother surfaces. This optimization have a tendency to rotate the object in 45 degrees in X and Y axes -- not exactly, but roughly, for a lot of objects.

Now there is a different heuristic which is trying to minimize the supports needed to print the object. I personally felt that this is more useful as the surface quality is also hugely dependent on the number of supports needed. So now you can't see an eye catching 45 degree rotation, but something that seems random, if pad around object is not enabled. It's not random however and it tries to minimize the needed supports.

With "pad around object" is enabled, the algorithm is extended to find the best surface of the object to put on the bed. If you want to do a print without elevation, I think there is little use of an auto rotation that will totally ignore the fact that you want to place the object directly on the bed.

nolspeed29, 2021-01-12:
Hmm, while I see where you are coming from with this change, often the main reason to orient at the angled rotation is to improve print success by “spreading out” the model across more supports and more of the build plate. This decreases the chances of the model detaching from supports or the build plate and leading to a failed print due to FEP film suction or adhesion. Also, angling the model can strengthen thin features, mask any faint layering, and reduce print time (less height than a vertical model). 

Could the option be put in to revert to the previous behavior?  What has the testing on the SL1 been like using this new auto orientation?

bubnikv, 2021-01-12:
@nolspeed29 We plan a minor release soon. Let's collect some feedback to this change and let's make a decision based on the feedback collected. As of now we have a feedback of just two of you.

dhprice, 2021-01-12:
I would second nolspeed29's suggestion.  I don't place objects flat on the build plate. I found the old optimisation worked very well with both complex shapes such as figurines and more regular objects such as vehicles. Not only did it result in a well supported object but also lower in height that the new method with shorter print times.  When I rotated and set supports manually I would get failures, with the old optimisation function I never had a print failure. Please reinstate the old method as an option.

TKIY, 2021-01-13:
Have to agree, the new orientation functionality is not at all workable for me.  This has completely killed my workflow and at the very least an option to choose behaviours is needed.

neophyl, 2021-01-13:
I don't use SLA personally so I don't really have an opinion but just to mention that this has been commented on the forum too https://forum.prusaprinters.org/forum/prusaslicer/prusaslicer-2-3-0-final-sla-optimise-orientation-behaviour-changed/
(for purposes of keeping track of feedback)

dhprice, 2021-01-13:
If you are looking for feedback you might cast your net a bit wider.  PrusaSlicer with the "old" orientation behaviour has been highly recommended for it's orientation and support generation for non-Prusa SLA printers, replacing those functions in Chitubox which is simply used to slice the result.

nolspeed29, 2021-01-14:
@dhprice @TKIY Nailed it.  This feature's behavior was so good in the previous releases that people use it for all SLA printers, not just Prusa SL1.  It is an integral part of the workflow.  Can an update be pushed to offer an option on the behavior?

TKIY, 2021-01-17:
Just tried orienting some pieces today after a configuration change push notification and this seems to have been reverted!

TKIY, 2021-01-19:
Not to be a pain, but any news on this being reverted or optionally reverted?

bubnikv, 2021-01-20:
We will likely do that in the upcoming minor release.

dhprice, 2021-04-16:
Any plans to re-introduce the ability to use the original orientation optimisation as it does not appear to have made it into in  2.3?

TKIY, 2021-04-16:
Still waiting for the 'minor release' and it's killing me.

This is still causing me a ton of extra unnecessary work.  No reason it should have been changed in the first place.

TKIY, 2021-04-24:
Not only was this not fixed in 2.3.1 it's now even worse.  You can no longer optimize the orientation of an object if ANY object is off the print bed.

PLEASE PLEASE PLEASE return the functionality that was removed for no apparent reason!

dhprice, 2021-04-24:
I like the Prusa SLicer for the Mk3 but have no other option as I cannot get 2.2 to co-exist successfully with later versions. So while I have a Prusa i3 Mk3 I have had to stop using Prusa Slicer for it.

Instead I have had to remove 3.3 and have reverted to PrusaSlicer 2.2 to generate pre-supports for a Creality SLA printer and now use Cura 4.9 for the Prusa Mk3.  

I am disappointed that the change in 2.3 appears to be solely because of the preference of one person and despite a number of customers/users requesting that the change be reverted or made optional for sound reasons we have been ignored.

crimmsonwind, 2021-05-05:
Piling on to say that I greatly preferred 2.2's Optimize Orientation as well. Prusaslicer has been frequently recommended on the various resin printing subreddits specifically because the Optimize Orientation (and auto supports) are very user-friendly (especially for beginners) and it _was_ superior to pretty much any other slicer. Totally loved recommending it for doing the setup and watching people who were previously struggling hard suddenly having a good amount of success. 

2.3's approach is a huge downgrade. While I understand the potential reasoning behind it, I don't agree with it. The number of supports alone aren't an issue. Trying to keep those supports off of important faces (the details and front half of a miniature for example) is way more important to me. As it currently is in 2.3, it's constantly putting the miniature face-down and covering it in supports which is exactly what I don't want.

At least in 2.2 I could select the optimization a few times to cycle between different approaches. With this, there's only one choice and it is face down.

Maybe you want to argue that 2.'3's take on optimization is better for certain things.

If you absolutely want to keep the current optimization, could you not add two options to toggle between? The current logic, and the previous one used in 2.2? You could mark them as one that prioritizes less supports (2.3's version), and one that prioritizes a better cross-section change layer to layer (2.2's version).

lukasmatena, 2021-05-06:
@tamasmeszaros might give more details, but as far as I know, the old behavior was brought back by 4293a68aaa2966a2e708de905594f312e45f72bd, with a possibility to choose. The change just did not make it into 2.3.1 which ended up as a minimal patch release of 2.3.0. It will be included in 2.4.0. You're not ignored, although I understand where the feeling came from.

[image]

dhprice, 2021-05-06:
Thanks for the update, look forward to trying functionality in 2.4

Area5142, 2021-05-06:
I can confirm that version_2.4.0-alpha0-270-g123c5af34 (today) has three orientation optimizations:
- Z axis only
- Least Support
- Best surface quality

[image] Z axis only

[image] Least Support

[image] Best surface quality

All are same cube, same angle of view - not sure why "Least support" has more support than "Best surface quality".

crimmsonwind, 2021-05-06:
This is very good to hear/see. I really look forward to trying out the different options and seeing how they work out.

tamasmeszaros, 2021-08-16:
Guys sorry for the long wait. The revert in the code was done a pretty long time ago, it's just that the next "proper" release is just about to happen. As @lukasmatena stated, 2.3.1 is a minor patch release. To make up for the long wait, you can now use the old rotation, the "new" one and an extra "Z axis only" rotation which only rotates the object to have a minimum bounding box in Z axis. Also, all the rotations should be a lot faster in the new release.

tamasmeszaros, 2021-08-16:
@Area5142 Please note that the "Least support" option uses a heuristic and does not call the support generator. It will try to orient the faces of the model so that the overall slopes are minimized. The support generator is much more intricate and looks for overhanging points, edges, etc...  It can happen that for such a simple model as the cube, the result is not apparent or useful.

Area5142, 2021-08-17:
@tamasmeszaros maybe the names should be changed to avoid confusion:

1. Old name **"Z axis only"**, new name **"Minimize print height"**
2. Old name **"Least Support"**, new name **"Minimize overall slopes"**
3. Keep name **"Best surface quality"**

1) Orient the model to have least z height for faster print time.
2) Orient model to have fewer slopes and less need for support.
3) Orient model to get the best surface quality (??).

bubnikv, 2021-09-22:
Implemented in PrusaSlicer 2.4.0-alpha1.
Closing.

closed 2021-09-22 by bubnikv