prusa3d/PrusaSlicer #5719: Can not start slicer on mac Bigsur

fangshihyuan, 2021-01-08:
### Version
PrusaSlicer-2.3.0-rc3+-202101081008

### Operating system type + version
MAC OS BIG SUR bêta 11.2 (20D5029f)

### Behavior
good day for everyone

my problem is the slicer can not start after i upgrade to BIG SUR, the error code is:

Time Awake Since Boot: 91000 seconds

System Integrity Protection: disabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [2554]

Application Specific Information:
*** CFEqual() called with NULL first argument ***

Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   com.apple.CoreFoundation      	0x00007fff2058c0cb CFEqual.cold.1 + 14
1   com.apple.CoreFoundation      	0x00007fff203e44e1 CFEqual + 693
2   com.prusa3d.slic3r/           	0x000000010f06b810 -[RemovableDriveManagerMM list_dev] + 528
3   com.prusa3d.slic3r/           	0x000000010f06bab5 Slic3r::GUI::RemovableDriveManager::list_devices(std::__1::vector<Slic3r::GUI::DriveData, std::__1::allocator<Slic3r::GUI::DriveData> >&) const + 69
4   com.prusa3d.slic3r/           	0x000000010ef034e9 Slic3r::GUI::RemovableDriveManager::update() + 105
5   com.prusa3d.slic3r/           	0x000000010eda0aff Slic3r::GUI::Plater::priv::priv(Slic3r::GUI::Plater*, Slic3r::GUI::MainFrame*) + 14303
6   com.prusa3d.slic3r/           	0x000000010edba854 Slic3r::GUI::Plater::Plater(wxWindow*, Slic3r::GUI::MainFrame*) + 180
7   com.prusa3d.slic3r/           	0x000000010ed6b83f Slic3r::GUI::MainFrame::init_tabpanel() + 255
8   com.prusa3d.slic3r/           	0x000000010ed6a910 Slic3r::GUI::MainFrame::MainFrame() + 2208
9   com.prusa3d.slic3r/           	0x000000010ed508bb Slic3r::GUI::GUI_App::on_init_inner() + 3851
10  com.prusa3d.slic3r/           	0x000000010ed4f96b Slic3r::GUI::GUI_App::OnInit() + 11
11  com.prusa3d.slic3r/           	0x000000010fa06516 wxApp::CallOnInit() + 422
12  com.prusa3d.slic3r/           	0x000000010fb956e5 wxEntry(int&, wchar_t**) + 197
13  com.prusa3d.slic3r/           	0x000000010fb9593c wxEntry(int&, char**) + 60
14  com.prusa3d.slic3r/           	0x000000010ed42c8b Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 251
15  com.prusa3d.slic3r/           	0x000000010e5a2041 Slic3r::CLI::run(int, char**) + 18801
16  com.prusa3d.slic3r/           	0x000000010e5a6970 main + 352
17  libdyld.dylib                 	0x00007fff2037e621 start + 1

can any help me?

thanks a lot

AKlionRU, 2021-01-11:
Most likely the problem is because you have a beta system, everything works fine for me on system 11.1 (20C69)

bubnikv, 2021-01-11:
Is it the new M1 ARM platform, or Intel?

AKlionRU, 2021-01-11:
[image]

fangshihyuan, 2021-01-11:
Thanks guys
yes, i am using beta OS (bêta 11.2 (20D5029f)) with intel i7.
maybe i will try downgrade to 11.1 (20C69).
[image]

bubnikv, 2021-04-14:
@fangshihyuan is it still a problem?

fangshihyuan, 2021-04-15:
Yes, it is still the same. I am trying to install the 2.3.1 alpha but it don’t work ether.
Do you have any idea for this ?

Thanks for your help.

---- St. MICHEL PATRON DES PARACHUTISTES PROTEGEZ NOUS DANS LE COMBAT ----

bubnikv, 2021-04-15:
Would you please try https://www.dropbox.com/s/pp4rski884fw5h7/PrusaSlicer-2.3.1-rc%2B4-universal-g506112422-202104151424.dmg?dl=0 ?

The crash log that you posted indicates an unexpected null value returned by the OSX API, which I may have fixed with 50611242253fa8a9237058bd84280f6025c8c8f8

fangshihyuan, 2021-04-16:
Dear sir

Thanks for your help,
Yes, it works so far but all the configuration is setting by default, I will keeping test it on My ender 3.
Thanks again

FANG

---- St. MICHEL PATRON DES PARACHUTISTES PROTEGEZ NOUS DANS LE COMBAT ----

bubnikv, 2021-04-19:
Thanks for heads up. Closing.

closed 2021-04-19 by bubnikv