prusa3d/PrusaSlicer #5722: 2.3.0-rc3 on Linux: Not linking to gtk3

bnavigator, 2021-01-09:
### Version
2.3.0-rc3

### Operating system type + version
openSUSE Tumbleweed Linux

### Behavior
building PrusaSlicer 2.3.0-rc3 fails with the following error:

```
[   36s] + /usr/bin/cmake /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.3.0-rc3/. '-GUnix Makefiles' -DCMAKE_INSTALL_PREFIX:PATH=/usr -DINCLUDE_INSTALL_DIR:PATH=/usr/include -DLIB_INSTALL_DIR:PATH=/usr/lib64 -DSYSCONF_INSTALL_DIR:PATH=/etc -DSHARE_INSTALL_PREFIX:PATH=/usr/share -DCMAKE_INSTALL_LIBDIR:PATH=lib64 -DCMAKE_INSTALL_LIBEXECDIR=/usr/libexec -DCMAKE_BUILD_TYPE=RelWithDebInfo '-DCMAKE_C_FLAGS=-O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g -DNDEBUG' '-DCMAKE_CXX_FLAGS=-O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g -DNDEBUG' '-DCMAKE_Fortran_FLAGS=-O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g -DNDEBUG' '-DCMAKE_EXE_LINKER_FLAGS=-flto=auto -Wl,--as-needed -Wl,--no-undefined -Wl,-z,now' '-DCMAKE_MODULE_LINKER_FLAGS=-flto=auto -Wl,--as-needed' '-DCMAKE_SHARED_LINKER_FLAGS=-flto=auto -Wl,--as-needed -Wl,--no-undefined -Wl,-z,now' -DLIB_SUFFIX=64 -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON -DBUILD_SHARED_LIBS:BOOL=ON -DBUILD_STATIC_LIBS:BOOL=OFF -DCMAKE_COLOR_MAKEFILE:BOOL=OFF -DCMAKE_INSTALL_DO_STRIP:BOOL=OFF -DCMAKE_MODULES_INSTALL_DIR=/usr/lib64/cmake/PrusaSlicer -DSLIC3R_FHS=1 -DSLIC3R_GTK=3
[   37s] -- The C compiler identification is GNU 10.2.1
[   37s] -- The CXX compiler identification is GNU 10.2.1
[   37s] -- Detecting C compiler ABI info
[   37s] -- Detecting C compiler ABI info - done
[   37s] -- Check for working C compiler: /usr/bin/cc - skipped
[   37s] -- Detecting C compile features
[   37s] -- Detecting C compile features - done
[   37s] -- Detecting CXX compiler ABI info
[   37s] -- Detecting CXX compiler ABI info - done
[   37s] -- Check for working CXX compiler: /usr/bin/c++ - skipped
[   37s] -- Detecting CXX compile features
[   37s] -- Detecting CXX compile features - done
[   37s] -- SLIC3R_ASAN: OFF
[   37s] -- SLIC3R_BUILD_SANDBOXES: OFF
[   37s] -- SLIC3R_BUILD_TESTS: ON
[   37s] -- SLIC3R_FHS: 1
[   37s] -- SLIC3R_GTK: 3
[   37s] -- SLIC3R_GUI: ON
[   37s] -- SLIC3R_MSVC_COMPILE_PARALLEL: ON
[   37s] -- SLIC3R_MSVC_PDB: ON
[   37s] -- SLIC3R_PCH: ON
[   37s] -- SLIC3R_PERL_XS: OFF
[   37s] -- SLIC3R_PROFILE: OFF
[   37s] -- SLIC3R_STATIC: OFF
[   37s] -- SLIC3R_WX_STABLE: OFF
[   37s] -- SLIC3R_STATIC: OFF
[   37s] -- CMAKE_PREFIX_PATH: (default)
[   37s] -- Found PkgConfig: /usr/bin/pkg-config (found version "1.7.3") 
[   37s] -- Looking for pthread.h
[   37s] -- Looking for pthread.h - found
[   37s] -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
[   37s] -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
[   37s] -- Check if compiler accepts -pthread
[   38s] -- Check if compiler accepts -pthread - yes
[   38s] -- Found Threads: TRUE  
[   38s] CMake Warning (dev) at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:426 (message):
[   38s]   The package name passed to `find_package_handle_standard_args` (DBUS) does
[   38s]   not match the name of the calling package (DBus).  This can lead to
[   38s]   problems in calling code that expects `find_package` result variables
[   38s]   (e.g., `_FOUND`) to follow a certain pattern.
[   38s] Call Stack (most recent call first):
[   38s]   cmake/modules/FindDBus.cmake:59 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
[   38s]   CMakeLists.txt:169 (find_package)
[   38s] This warning is for project developers.  Use -Wno-dev to suppress it.
[   38s] 
[   38s] -- Found DBUS: /usr/include/dbus-1.0;/usr/lib64/dbus-1.0/include  
[   38s] -- Found Boost: /usr/lib64/cmake/Boost-1.75.0/BoostConfig.cmake (found suitable version "1.75.0", minimum required is "1.64.0") found components: system filesystem thread log locale regex chrono atomic date_time 
[   38s] -- Boost::boost exists
[   38s] -- Found TBB: /usr/include (found version "2020.3")  
[   38s] --   TBB_FOUND               = TRUE
[   38s] --   TBB_INCLUDE_DIRS        = /usr/include
[   38s] --   TBB_DEFINITIONS         = 
[   38s] --   TBB_LIBRARIES           = /usr/lib64/libtbb.so
[   38s] --   TBB_DEFINITIONS_DEBUG   = TBB_USE_DEBUG=1
[   38s] --   TBB_LIBRARIES_DEBUG     = 
[   38s] --   TBB_DEFINITIONS_RELEASE = 
[   38s] --   TBB_LIBRARIES_RELEASE   = /usr/lib64/libtbb.so
[   38s] -- Found CURL: /usr/lib64/libcurl.so (found version "7.74.0") 
[   38s] -- Found ZLIB: /usr/lib64/libz.so (found version "1.2.11") 
[... 129 lines cut ...]
[   40s]     SYSCONF_INSTALL_DIR
[   40s] 
[   40s] 
[   40s] -- Build files have been written to: /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.3.0-rc3/build
...
[  665s] [100%] Linking CXX executable prusa-slicer
[  665s] cd /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.3.0-rc3/build/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/PrusaSlicer.dir/link.txt --verbose=1
[  665s] /usr/bin/c++ -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g -DNDEBUG -fext-numeric-literals -Wall -Wno-reorder -pthread -O2 -g -DNDEBUG -flto=auto -Wl,--as-needed -Wl,--no-undefined -Wl,-z,now CMakeFiles/PrusaSlicer.dir/PrusaSlicer.cpp.o -o prusa-slicer  libslic3r/liblibslic3r.a -ldl -lstdc++ slic3r/liblibslic3r_gui.a -ldl libslic3r/liblibslic3r.a libnest2d/liblibnest2d.a /usr/lib64/libnlopt.so admesh/libadmesh.a miniz/libminiz_static.a /usr/lib64/libboost_log.so.1.75.0 /usr/lib64/libboost_filesystem.so.1.75.0 -lrt /usr/lib64/libboost_locale.so.1.75.0 /usr/lib64/libboost_thread.so.1.75.0 /usr/lib64/libboost_regex.so.1.75.0 -licudata -licui18n -licuuc /usr/lib64/libboost_chrono.so.1.75.0 /usr/lib64/libboost_atomic.so.1.75.0 /usr/lib64/libboost_date_time.so.1.75.0 clipper/libclipper.a boost/libnowide.a glu-libtess/libglu-libtess.a polypartition/libpolypartition.a poly2tri/libpoly2tri.a qhull/libqhullstatic.a -lm semver/libsemver.a libslic3r/liblibslic3r_cgal.a /usr/lib64/libCGAL.so.13.0.3 /usr/lib64/libgmpxx.so /usr/lib64/libmpfr.so /usr/lib64/libgmp.so /usr/lib64/libpng.so /usr/lib64/libopenvdb.so /usr/lib64/libtbb.so /usr/lib64/libboost_system.so.1.75.0 /usr/lib64/libboost_iostreams.so.1.75.0 -lbz2 -lz /usr/lib64/libHalf-2_5.so.25.0.2 -pthread avrdude/libavrdude.a imgui/libimgui.a /usr/lib64/libGLEW.so /usr/lib64/libGLU.so /usr/lib64/libGL.so hidapi/libhidapi.a -ldl /usr/lib64/libz.so /usr/lib64/libcurl.so -L/usr/lib64 -lwx_baseu-3.1 -lwx_gtk3u_core-3.1 -lwx_gtk3u_html-3.1 -lwx_gtk3u_gl-3.1 /usr/lib64/libpng.so /usr/lib64/libz.so /usr/lib64/libexpat.so /usr/lib64/librt.so /usr/lib64/libdbus-1.so 
[  665s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.3.0-rc3/src/avrdude/config.c:354: warning: type of 'yy_scan_bytes' does not match original declaration [-Wlto-type-mismatch]
[  665s]   354 | extern YY_BUFFER_STATE yy_scan_bytes(const char *base, size_t size);
[  665s]       | 
[  665s] avrdude/lexer.c:2825: note: type mismatch in parameter 2
[  665s] avrdude/lexer.c:2825: note: type 'int' should match type 'size_t'
[  665s] avrdude/lexer.c:2825: note: 'yy_scan_bytes' was previously declared here
[  665s] avrdude/lexer.c:2825: note: code may be misoptimized unless '-fno-strict-aliasing' is used
[  665s] /usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/prusa-slicer.tpSoK2.ltrans61.ltrans.o: undefined reference to symbol 'gtk_cell_layout_get_cells'
[  665s] /usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /usr/lib64/libgtk-3.so.0: error adding symbols: DSO missing from command line
[  665s] collect2: error: ld returned 1 exit status
```

Full build log with all package versions:  [PrusaSlicer-2.3.0-rc3_log.txt](https://github.com/prusa3d/PrusaSlicer/files/5790904/PrusaSlicer-2.3.0-rc3_log.txt)

2.3.0-rc2 built fine.

Did you forget something in the CMake files, or do I need to adjust my build recipe? Or is it a bug in the openSUSE gtk3 package?

fiechr, 2021-01-10:
Same error on Manjaro/Arch. All previous beta and RCs built fine.

```
[...]
/usr/bin/ld: slic3r/liblibslic3r_gui.a(PresetComboBoxes.cpp.o): undefined reference to symbol 'g_type_check_instance_cast'
/usr/bin/ld: /usr/lib/libgobject-2.0.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
```

fiechr, 2021-01-10:
It may be related to https://github.com/prusa3d/PrusaSlicer/commit/2505c0a82dcbb1ec8f607218a336c4a16b7c03b2.

bnavigator, 2021-01-10:
Looks **very** related.

clytle374, 2021-01-10:
I'm having the same build failure, jumping back one commit to a5882a1a1c3f561233a22cc7663514b97e5fb3ba  and something else broke
then tried 
efd3002ea9c5c9f76e94686ea77f9c49318654b4
and it fixed the issue.
Thanks, Cory

combolek, 2021-01-10:
Can you try PR #5717 ? This may fix it.

combolek, 2021-01-10:
Never mind, I just looked more closely and yeah, this is the same error.

bnavigator, 2021-01-10:
Can confirm, #5717 fixes it: [PrusaSlicer-2.3.0-rc3_fixed_log.txt](https://github.com/prusa3d/PrusaSlicer/files/5793250/PrusaSlicer-2.3.0-rc3_fixed_log.txt)

I didn't find your PR when I search for gtk3 or linking issues before opening this issue. A more descriptive explanation might have helped. Also, it is not *exactly* the same error. We have three different undefined symbol references, but all in the same library package.

commit b7bfaea1 (2021-01-11): Merge pull request #5717 from combolek/master

Fixed build regression in commit a5882a1a1c3f561233a22cc7663514b97e5f…
fixes #5722
fixes #5739
closed 2021-01-11 by tamasmeszaros