prusa3d/PrusaSlicer #5725: Supports in long Gaps with "Don't support Bridges disabled" not "Support from Build Plate only" disabled

IteDas, 2021-01-09:
### Version
2.3.0-rc3

### Operating system type + version
Windows 10-x64 

### 3D printer brand / version + firmware version (if known)
Prusa MK3s 3.92

### Behavior
* Supports on slots that are correctly detected as bridges can not be support unless they support from Build Plate 
* Load STL with a long gap
* Slice with "Don't support Bridges" disabled and "Generate Supports" enabled and either "Auto-Generate Supports" or draw Support Enforcers
* 
[image]

* _Expected Results_
* correctly supported bridges
* Simplify 3D Result:
[image]

* _Actual Results_
  * no bridge support
  * PrusaSlicer Result:
[image]

_Is this a new feature request?_
No

mansson, 2021-01-09:
Same on PrusaSlicer-2.3.0-rc3+linux-x64-202101081026.AppImage. Surely this behaviour was not intended? Using all default settings, support will only appear if I "spill over" on the vertical surface, then support will go down to the plate from this point. All areas where I have painted on the horizontal underside of a bridged structure, there will be no support generated down to the underlying surface. Painting, but not getting any support from the visibly painted area is an unexpected behaviour. It's not mentioned in the instruction video that this can happen.

bubnikv, 2021-10-05:
There is no 3MF provided, but with PrusaSlicer 2.4.0-alpha3 snug supports the issue is resolved.

closed 2021-10-05 by bubnikv