prusa3d/PrusaSlicer #5727: No error when uploading print to Printer Host when printer not connected

RichiH, 2021-01-09:
### Version
2.3.0 RC3, Linux

### Operating system type + version
Debian unstable/testing

### 3D printer brand / version + firmware version (if known)
Prusa Mini+, 4.2.1
Octoprint 1.5.2

### Behavior
* Disconnect Octoprint from Prusa Mini in software (e.g. to change filament), USB connection is fine 
* Slice a model
* Click the "Send to printer"
* Enable "Start printing after upload"
* GCode transmits correctly, print does NOT start and NO warning is shown

I would expect an error/warning instead.

I don't know if this is an issue with Prusa Slicer, Octoprint, or both, but I have to start somewhere so I did it here :)

[image]

bubnikv, 2021-01-11:
Do you get a "failed" message in the upload log?

Go to menu Window -> Print Host Upload Queue.

We should add a notification for "upload finished" / "upload failed".

RichiH, 2021-01-11:
I can check after the current print finishes.

I would argue that "success" is optional and it would be enough to have one
of those new-style overlay notifications in the lower right of the 3D view.
For "error" I would strongly encourage a pop-up so it can't be missed when
you're in a hurry.

>

RichiH, 2021-01-12:
I disconnected OctoPi from my printer, sent a print job, and:

[image]

RichiH, 2021-01-12:
I disconnected Octopi from my printer, sent a job, and:

[image]

bubnikv, 2021-01-12:
Completed means transferred, not printed.

RichiH, 2021-01-12:
Arguably that should be renamed then.

Yet, the main bug is that PrusaSlicer does not inform the user that the print didn't, in fact, start.

bubnikv, 2021-01-13:
> Yet, the main bug is that PrusaSlicer does not inform the user that the
print didn't, in fact, start.

I am not sure whether the OctoPrint API supports that, at least in the
upload API. I suppose PrusaSlicer would have to invoke another API,
possibly after a while.

RichiH, 2021-01-13:
That's absolutely possible; I had to start somewhere, but I can create a sibling issue in Octoprint, link the two, and try to facilitate the conversation that way.

closed 2026-09-29 by prusaslicer-bot[bot]