prusa3d/PrusaSlicer #5733: Lock files in the local configuration directory are not deleted

aegean-odyssey, 2021-01-10:
### Version

Version 2.3.0-rc3+linux-x64

### Operating system type + version

PrusaSlicer-2.3.0-rc3+linux-x64-202101081026.AppImage
PrusaSlicer-2.3.0-rc2+linux-x64-GTK3-202012221657.AppImage

### Behavior

I am not sure if this issue affects other platforms, but on linux the lockfiles created in the config directory, `PrusaSlicer/cache/`, are being created but not deleted. With each invocation of the PrusaSlicer application, the number of "orphaned" lockfiles increases. I think these files should be deleted when they are no longer needed by the application.

combolek, 2021-01-11:
Does this happen when you use the tarball ([PrusaSlicer-2.3.0-rc3+linux-x64-202101081026.tar.bz2](https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.3.0-rc3/PrusaSlicer-2.3.0-rc3+linux-x64-202101081026.tar.bz2)) instead of the AppImage?

The lock file name is derived from the full path to the prusa-slicer executable. Normally when the app is installed in a fixed location the executable path will be always the same, so you should only have one such lock file. But AppImage is mounted under a different temporary directory each time you run it, so the path and hence the lock file name is different each time.

So actually the bigger problem is that the `--single-instance` option is broken with AppImages, because each instance is using a different lock file which defeats their purpose.

combolek, 2021-01-11:
I would suggest that this issue is renamed to something like "Single instance mode does not work with AppImage".

bubnikv, 2021-01-11:
It should be fixed with https://github.com/prusa3d/PrusaSlicer/commit/7780221683adca49e3e336c38be82134ee8422fd
by taking the path to the AppImage instead of the path to the PrusaSlicer binary.

closed 2021-01-11 by bubnikv

aegean-odyssey, 2021-01-13:
Great, you guys are sharp.