prusa3d/PrusaSlicer #5734: SD Card Ejection Problem - Ejection of Device Failed Issue

sarvenn, 2021-01-10:
### Version
2.3.0 rc3 win64

### Operating system type + version
Win10 64-bit

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro SKR Mini E3 V2 + TFT35 E3 V3 Marlin 2.0.7

### Behavior
Whenever I export a gcode to my SD card and click on eject button it is unsuccessfull with  "ejection of device F: failed" messege.
In Cura I had no problem with in app ejection feature.

bubnikv, 2021-01-11:
Don't you have another application running, that possibly locks the SD card? Maybe Cura blocks it?

sarvenn, 2021-01-11:
@bubnikv No. Only prusaslicer was open. After I get this messege inside prusaliscer, I always could eject the SD card from taskbar icon. That's why I'm sure that another app is not interrupting.

sarvenn, 2021-01-12:
@bubnikv  I installed the final release v2.3 from Prusa website and tried again.
Issue still exists.

sarvenn, 2021-01-19:
I also can eject the SD card in Cura as well.
I never could successfully eject SD card in PS.

sarvenn, 2021-01-31:
Help please? No response for 20 days.

sarvenn, 2021-12-29:
Problem still continues in 2.4.
As I told before nothing is using the SD card because I can eject my SD card with taskbar icon without any problem.
I also could eject my SD card inside Cura. But inside PrusaSlicer, I always get error messege.

sarvenn, 2021-12-29:
@bubnikv 
Here is a screenshot of the issue:
[image]

My system info:
PrusaSlicer
Version:   2.4.0-beta1+win64
Build:     PrusaSlicer-2.4.0-beta1+win64-202110251333

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 4,293MB
OpenGL installation
GL version:   3.3.11672 Compatibility Profile Context
Vendor:       ATI Technologies Inc.
Renderer:     ATI Mobility Radeon HD 3650
GLSL version: 3.30
<details>
<summary>Installed extensions:</summary>
GL_AMDX_debug_output
GL_AMDX_vertex_shader_tessellator
GL_AMD_conservative_depth
GL_AMD_debug_output
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_name_gen_delete
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_sample_positions
GL_AMD_shader_stencil_export
GL_AMD_vertex_shader_tessellator
GL_ARB_ES2_compatibility
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_objects
GL_ARB_shader_precision
GL_ARB_shader_stencil_export
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_snorm
GL_ARB_texture_storage
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ARB_window_pos
GL_ATI_draw_buffers
GL_ATI_envmap_bumpmap
GL_ATI_fragment_shader
GL_ATI_meminfo
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_copy_buffer
GL_EXT_copy_texture
GL_EXT_direct_state_access
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_histogram
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texgen_reflection
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_buffer_object
GL_EXT_texture_compression_latc
GL_EXT_texture_compression_rgtc
GL_EXT_texture_compression_s3tc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_IBM_texture_mirrored_repeat
GL_KTX_buffer_region
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_explicit_multisample
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

lukasmatena, 2021-12-29:
This reminds me of #3975. Can you please try to change the label to something without spaces? Does the problem happen with any SD card?

sarvenn, 2021-12-29:
> This reminds me of #3975. Can you please try to change the label to something without spaces? Does the problem happen with any SD card?

Hi @lukasmatena,
I have tried with 3 different SD cards. One had no naming/label, one had  single word label, other had two word label.
All of them have the same warning result and failure. All of them can be successfully ejected from task bar icon, remove disk selection. I'm totally sure that nothing is using the SD card during this warning in PS.

sarvenn, 2022-01-01:
@bubnikv ,
Can't you fix the issue above with another pull request, just like you solved for Linux? #5734

bubnikv, 2022-01-03:
I would fix it if I knew how.
What is your SD card reader? I think we have had a single SD card reader with similar issue, but on very most Windows computers we don't have your issue.

sarvenn, 2022-01-03:
> I would fix it if I knew how. What is your SD card reader? I think we have had a single SD card reader with similar issue, but on very most Windows computers we don't have your issue.

It's an onboard SD card reader on my Toshiba Satellite A300-1ND laptop. It's an old model but still working :)
I googled it and found O2MİCRO CARD READER. I couldn't find the name of the OEM in my device manager.
If it is spesific to me and doesn't have an easy solution just tell me I can close the issue, it's not a big issue.

bubnikv, 2022-01-03:
> If it is spesific to me and doesn't have an easy solution just tell me I can close the issue, it's not a big issue.

You can close this issue or leave it open for others to chime in. However usually we are unable to fix such issues if we cannot get our hands on a similar hardware. And if the issue is extremely rare, we most likely not fix the issue for you.

closed 2022-01-04 by sarvenn