prusa3d/PrusaSlicer #5735: Free run of the extruder. Dirt when printing on the inner diameter surface.

AKlionRU, 2021-01-10:
Forgive me, I do not know English well, I am from Russia, therefore I am writing through google translator.

When printing parts to a new printer, I noticed that the parts on the outside are of perfect quality, but the inner diameters are not printed cleanly. I think this is due to the fact that the idle of the printer goes straight along the edge of the inner diameter. 
[image]

Is it possible to make it so that the idle line is 1-3 perimeters from the edge of the model, and so that this distance can be chosen?
[image]
If I am wrong then correct me. Maybe the reason for the not clean print is different? But I set up the printer perfectly, all the outer surfaces are just perfect.

Area5142, 2021-01-11:
Please look at issue #5689, I think it is a description of the same problem.

AKlionRU, 2021-01-11:
Yes, this is a description of the same problem, but there is no solution to the problem. I have already tried the settings to solve this problem, but it is not solved simply by the settings. I have been trying to solve this problem for a long time, but I still can't find a solution. In general, I have been doing printing for almost 10 years, and for almost 10 years this problem comes up periodically.

AKlionRU, 2021-01-11:
I still think that it is necessary to correct the algorithm of the program, and this is the only way to fix this problem.

JoeMudr, 2021-02-22:
The only thing i found that helped a bit was disabling "only retract when crossing perimeters". But this for some reason creates a hole, where it starts the travel.

closed 2026-09-29 by prusaslicer-bot[bot]