prusa3d/PrusaSlicer #5737: park move scar on ironed surface

Moyrn, 2021-01-10:
### Version
2.3 RC3 Win64

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMU2S (3.9.1)

### Behavior
* _Describe the problem_
After printing a surface flat, then iron the top surface
the last movement ruins the nice surface with a move scar

From the GCODE
G1 Z39.95 ; Move print head up
G1 X0 Y200 F3000 ; home X axis

should perhaps be
G1 Z45 ; Move print head up

* _Steps needed to reproduce the problem_
* iron

  * _If this is a command-line slicing issue, include the options used_
NO
 
* _Expected Results_
NO scar after ironing

* _Actual Results_
park move scar after ironing

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_
NO
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-01-10:
Please provide your 3mf project file.

Moyrn, 2021-01-11:
Sorry for my newbeetude. How do I attach a file here. I tried drag & drop of the 3MF, renaming it as Zip... And I do not see a "file attach" option in the buttons-bar. Thanks.

Moyrn, 2021-01-11:
[poseidon-trident-sigma-v2-nologo.zip](https://github.com/prusa3d/PrusaSlicer/files/5794414/poseidon-trident-sigma-v2-nologo.zip)

>> ok my previous tries were with a 7z archive type. Read too fast. Here's the 3MF zipped. thanks for your kind attention

Note: I tried the Z45 but not better - perhaps too slow. Now i try to park away not passing over the printed area.

Moyrn, 2021-01-11:
It seems the end-code 
G1 Z39.95 ; Move print head up

Should wait completion to proceed to next step. Else the subsequent G1 X0 Y200 starts too early / fast and then issues a scar on the surface it flies over

bubnikv, 2021-01-11:
> It seems the end-code
> G1 Z39.95 ; Move print head up

> Should wait completion to proceed to next step. Else the subsequent G1 X0
Y200 starts too early / fast and then issues a scar on the surface it flies
over

I don't understand. The firmware does not reorder G-code lines.

Moyrn, 2021-01-11:
it was just my perception. Despite the move up Z, the scar is still present. as if the G1 Z was either ignored, ot just launched asynchronously. Because the G1 Z45 issued the same scar :-(
You have the M3F attached. I added a result photo as well
[image]

Moyrn, 2021-01-11:
Ok i think i found the issue. This is because of the wipe tower

{if has_wipe_tower}
G1 E-15.0000 F3000
{else}
G1 X0 Y210 F7200

this is issued BEFORE the lines I mentionned earlier. My apologies...

bubnikv, 2021-01-11:
The issue may be caused by the custom end G-code for MMU2 profiles if the wipe tower is not active.

```
{if has_wipe_tower}
G1 E-15.0000 F3000
{else}
G1 X0 Y210 F7200
G1 E2 F5000
G1 E2 F5500
G1 E2 F6000
G1 E-15.0000 F5800
G1 E-20.0000 F5500
G1 E10.0000 F3000
G1 E-10.0000 F3100
G1 E10.0000 F3150
G1 E-10.0000 F3250
G1 E10.0000 F3300
{endif}

; Unload filament
M702 C

G4 ; wait
M221 S100 ; reset flow
M900 K0 ; reset LA
{if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|SOLUBLE|@0.25 nozzle MK3).*/}M907 E538 ; reset extruder motor current{endif}
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
; Lift print head a bit
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+30, max_print_height)}{endif} ; Move print head up
G1 X0 Y200 F3000 ; home X axis
M84 ; disable motors
```

Moyrn, 2021-01-11:
Yes. I just noticed that (see my previous post)

bubnikv, 2021-01-11:
Thanks for your findings, it is an issue of your MMU2 profiles if printing with a single extruder only.

rtyr, 2021-01-11:
I will update the profiles to prevent this.

Moyrn, 2021-01-11:
Many thanks :-)

rtyr, 2021-01-14:
Updated in the configuration update 1.2.3. I also changed the location of the `M140 S0 ; turn off heatbed ` command. It is now placed before final unload (M702 C) to prevent unnecessary heating of the bed if the unload fails for some reason ("MMU needs attention" state). Closing.

closed 2021-01-14 by rtyr

Moyrn, 2021-01-14:
Many thanks - Dekuji !