prusa3d/PrusaSlicer #5738: Temperature change on layer change does not work.

MeisterQ, 2021-01-10:
### Version
2.3.0-rc3

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### 3D printer brand / version + firmware version (if known)
Hypercube Evolution. Marlin 2.0.x 

### Behavior
I want to create a heating tower. So i used the custom code, which is providet at the prusa page:
{if layer_z < 42}M104 S260
{elsif layer_z < 76}M104 S255
{elsif layer_z < 110}M104 S250
{elsif layer_z < 144}M104 S245
{elsif layer_z < 178}M104 S240
{elsif layer_z < 212}M104 S235
{elsif layer_z < 246}M104 S230
{elsif layer_z < 280}M104 S225
{elsif layer_z > 280}M104 S220
{endif}

After creating the gcode and start to print i noticed, that the temperature wasnt changed after the specified layers.
So i tried to manually change the temperature, but it didnt let me. It always changed the temperature back to the starting temp.

After inspecting the .gcode file i figured out, that the M104 S260 command is written in the code very often.
I attached the gcode file. 
[220-260PETG_Temp_Tower_.txt](https://github.com/prusa3d/PrusaSlicer/files/5793267/220-260PETG_Temp_Tower_.txt)

I also put before the custom gcode the commands "G1 X0 F5000" and it was executed on every layer change. Just the temperature didnt.

So i tried it was an old Slicer, like the Slic3r Prusa Edition. There, in the gcode, my custom gcode was written and the print worked.

Any ideas? Seems like a bug.

rtyr, 2021-01-10:
Your layer_z values are incorrect (layer numbers instead of Z positions).

MeisterQ, 2021-01-11:
> 
> 
> Your layer_z values are incorrect (layer numbers instead of Z positions).

It is the layernumber according to what Prusaslicer is showing me on the right.

bubnikv, 2021-01-11:
exactly, but layer_z is a height in mm, not index of a layer.

MeisterQ, 2021-01-11:
> 
> […](#)

Thank you. So i totally missunderstood the explanation at prusa page.

I will try to use the height in millimeters and try to print again.

So now i also understand why its sending the M104 command all the time, because its sending it even on Z-Hop?.

rtyr, 2021-01-14:
> So now i also understand why its sending the M104 command all the time, because its sending it even on Z-Hop?.

No, it just adds M104 Sxxx to all layers, which meets the condition. For example `{if layer_z < 42}M104 S260` will add M104 S260 to all layers below 42mm. You can change `<` to `==` if you want M104 commands only at specific layers, but it doesn't matter in this particular case (both variants will work fine).

Can we close this issue?

MeisterQ, 2021-01-14:
I solved my problem with the height in Millimeters yes.

So thank you all and it can be closed

closed 2021-01-14 by MeisterQ