prusa3d/PrusaSlicer #5743: BUG: cannot add more than one height range modifier

amper128, 2021-01-11:
### Version
PrusaSlicer-2.3.0-rc3+linux-x64-GTK3-202101081058

### Operating system type + version
Linux

### Behavior
I cant add multiple height ranges modifier for part. In Release 2.2 it was worked.

[image]

Sometimes i see "minus" icon, but no "plus" to add new range.

bubnikv, 2021-01-11:
The plus / minus buttons are accessible when you select the "Layers" item just 1 above the selected item in the Object list.

YuSanka, 2021-01-11:
Try to select "Layers" ("Слои") :)
[image]

bubnikv, 2021-01-11:
[image]

amper128, 2021-01-11:
[image]
Only minus button. No plus.

YuSanka, 2021-01-11:
> Only minus button. No plus.

Ok, you are right. This issue will be fixed with updated localization

YuSanka, 2021-06-04:
This bug is fixed and will be a part of upcoming release.

closed 2021-06-04 by YuSanka

bubnikv, 2021-08-06:
Russian dictionaries were updated in PrusaSlicer 2.3.3, thus this bug should already be fixed there.