prusa3d/PrusaSlicer #5744: Live adjust Z while ironing

mindelheini, 2021-01-11:
VERSION 2.3

WINDOWS 10

PRUSA I3 MK3S+    fw 3.9.3-3556
### Behavior
It would be good to have the live adjust z in the setting avaiable to tune the nozzle height while the printer is ironing.
That's currently that's not avaiable 

I printed a 6mm tall object with a 0.6 mm nozzle and the surface was not perfectly even

#5743 
[image]

rtyr, 2021-01-11:
It would certainly not help. It looks like partial clog (likely heat creep related).

fe60, 2021-01-15:
That's indeed caused by a clogged nozzle. Unfortunately I know very well how such prints look like.
https://forum.prusaprinters.org/forum/prusaslicer/prusaslicer-2-3-0-beta2-ironing-causes-nozzle-jam/

mindelheini, 2021-01-15:
The problem was that I used not enough infill and the surface was not
perfectly flat

You are right it would not help to have live z adjust while ironing

rtyr, 2021-01-15:
OK, I am closing this issue.

closed 2021-01-15 by rtyr