prusa3d/PrusaSlicer #5745: Crash when slicing instanced object

smirgol, 2021-01-11:
### Version
2.3.0+linux-x64

### Operating system type + version
Linux - Ubuntu 20.04

### 3D printer brand / version + firmware version (if known)
Prusa MK3s+

### Behavior
* Download foot.stl from [Thingiverse](https://www.thingiverse.com/thing:2841380/files) (other objects will do as well)
* Open PrusaSlicer and drag STL file on printing bed.
* Add 3 copies
* Auto arrange
* Click Slice
* It will slice but then crash

_Expected Results_
Not crashing

_Actual Results_
Crashing

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[foot.zip](https://github.com/prusa3d/PrusaSlicer/files/5798456/foot.zip)

#### Additional comment
For this specific object, it will only crash when creating at least 3 additional instances (total of 4 objects). It will not crash with fewer objects.
For other STLs though, it might also crash with fewer instances, but rarely at 2.

lukasmatena, 2021-01-12:
I cannot reproduce it (Ubuntu 16.04). Can you please run the application from command line with `--loglevel=5` argument if it gives a clue about what is wrong? Are you able to provide any other debugging info (such as a gdb backtrace)? Thanks.

smirgol, 2021-01-12:
Uhm I unfortunately don't have much experience with gdb. But when it crashed this is what it said:
```
Thread 1 "slic3r_main" received signal SIGSEGV, Segmentation fault.
0x00007fffec677299 in ?? () from /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
```

Then I've typed `backtrace` and this is the result:
```
#0  0x00007fffec677299 in  () at /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
#1  0x00007fffec679bcb in  () at /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
#2  0x00007fffecc47a36 in  () at /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
#3  0x00007fffed1208c9 in  () at /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
#4  0x00007fffed120c88 in  () at /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
#5  0x00007fffed13b41a in  () at /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
#6  0x00007fffed05fd25 in  () at /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
#7  0x00007fffed069fb2 in  () at /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
#8  0x0000000000c9a7e0 in Slic3r::GUI::GCodeViewer::render_toolpaths() const ()
#9  0x0000000000ca31a5 in Slic3r::GUI::GCodeViewer::render() const ()
#10 0x0000000000c2a14b in Slic3r::GUI::GLCanvas3D::render() ()
#11 0x0000000000cbfaf3 in Slic3r::GUI::Preview::on_moves_slider_scroll_changed(wxCommandEvent&) ()
#12 0x000000000182e1be in wxAppConsoleBase::HandleEvent(wxEvtHandler*, void (wxEvtHandler::*)(wxEvent&), wxEvent&) const (this=0x237d350, handler=0x2f4a400, func=
    (void (wxEvtHandler::*)(class wxEvtHandler * const, class wxEvent &)) 0xcbfa70 <Slic3r::GUI::Preview::on_moves_slider_scroll_changed(wxCommandEvent&)>, event=...)
    at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appbase.cpp:657
#13 0x000000000182e232 in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const (this=0x237d350, handler=0x2f4a400, functor=Warnung: RTTI symbol not found for class 'wxEventFunctorMethod<wxEventTypeTag<wxScrollEvent>, Slic3r::GUI::Preview, wxCommandEvent, Slic3r::GUI::Preview>'
..., event=...)
    at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appbase.cpp:669
#14 0x00000000018c2906 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (entry=..., handler=0x2f4a400, event=...)
    at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1415
#15 0x00000000018c3567 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) (this=0x2d1cb90, event=...) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1887
#16 0x00000000018c2d2a in wxEvtHandler::TryHereOnly(wxEvent&) (this=0x2d1cb90, event=...) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1608
#17 0x00000000018c40bb in wxEvtHandler::TryBeforeAndHere(wxEvent&) (this=0x2d1cb90, event=...) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/include/wx/event.h:3912
#18 0x00000000018c2b9b in wxEvtHandler::ProcessEventLocally(wxEvent&) (this=0x2d1cb90, event=...) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1545
#19 0x00000000018c2b32 in wxEvtHandler::ProcessEvent(wxEvent&) (this=0x2d1cb90, event=...) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1518
#20 0x0000000000cd5b6b in Slic3r::DoubleSlider::Control::SetSelectionSpan(int, int) ()
#21 0x0000000000cc0c0a in Slic3r::GUI::Preview::update_moves_slider() ()
#22 0x0000000000cc3edd in Slic3r::GUI::Preview::on_layers_slider_scroll_changed(wxCommandEvent&) [clone .part.622] ()
#23 0x000000000182e1be in wxAppConsoleBase::HandleEvent(wxEvtHandler*, void (wxEvtHandler::*)(wxEvent&), wxEvent&) const (this=0x237d350, handler=0x2f4a400, func=
    (void (wxEvtHandler::*)(class wxEvtHandler * const, class wxEvent &)) 0xcc4090 <Slic3r::GUI::Preview::on_layers_slider_scroll_changed(wxCommandEvent&)>, event=...)
    at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appbase.cpp:657
#24 0x000000000182e232 in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const (this=0x237d350, handler=0x2f4a400, functor=Warnung: RTTI symbol not found for class 'wxEventFunctorMethod<wxEventTypeTag<wxScrollEvent>, Slic3r::GUI::Preview, wxCommandEvent, Slic3r::GUI::Preview>'
..., event=...)
    at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appbase.cpp:669
#25 0x00000000018c2906 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (entry=..., handler=0x2f4a400, event=...)
    at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1415
#26 0x00000000018c3567 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) (this=0x2d66380, event=...) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1887
#27 0x00000000018c2d2a in wxEvtHandler::TryHereOnly(wxEvent&) (this=0x2d66380, event=...) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1608
#28 0x00000000018c40bb in wxEvtHandler::TryBeforeAndHere(wxEvent&) (this=0x2d66380, event=...) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/include/wx/event.h:3912
[... cut ...]
..., event=Warnung: RTTI symbol not found for class 'Slic3r::SlicingProcessCompletedEvent'
...)

```

I'm using an AMD RX5700XT card and the latest oibaf mesa drivers, if that matters.

As for the terminal output, I can't see anything noteable here. But I've attached it anyways. Strange thing is,  when redirecting the output it will stop in the middle of "Exported leyer 88", but when not redirecting the output, I get more lines in the terminal - I've attached those additional lines to the log:
[logfile.txt](https://github.com/prusa3d/PrusaSlicer/files/5799064/logfile.txt)

bubnikv, 2021-01-12:
The call stack indicates, that it crashes in "render_toolpaths()"
Are you adding the instances and doing the arrange in the Plater or in
G-code preview? (I am not sure how could you do the operations in G-code
preview, but sometimes customers surprise us)?

smirgol, 2021-01-12:
No, nothing special, really. I'm in expert mode, drag the STL on the bed, hit the "+" button at the top three times to create 3 additional instances, click the auto-arrange button and then hit slice. It slices, generates the G-Code, but then it crashes. But only when I have more than 2 instances (3 objects in total). This differs if I use other STLs, but for this specific one the magic number is 3 objects, anything above will crash.
And it will also crash if I just drag the STL to the bed 4 times and arrange manually, instead of using the buttons.

smirgol, 2021-01-16:
I managed to build a debug version and here's a backtrace with some more info (top 10 lines):

```
Thread 1 "slic3r_main" received signal SIGSEGV, Segmentation fault.
0x00007fffecc85ee9 in ?? () from /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
(gdb) backtrace
#0  0x00007fffecc85ee9 in  () at /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
#1  0x00007fffecc8824b in  () at /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
#2  0x00007fffed74f026 in  () at /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
#3  0x00007fffed74f3d8 in  () at /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
#4  0x00007fffed769daa in  () at /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
#5  0x00007fffed68e5a5 in  () at /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
#6  0x00007fffed698922 in  () at /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
#7  0x000055555697524a in Slic3r::GUI::GCodeViewer::<lambda(const Slic3r::GUI::GCodeViewer::TBuffer&, unsigned int, Slic3r::GLShaderProgram&)>::operator()(const Slic3r::GUI::GCodeViewer::TBuffer &, unsigned int, Slic3r::GLShaderProgram &) const (__closure=0x7fffffff8190, buffer=..., index_buffer_id=0, shader=...) at /media/Gonzales/compile/PrusaSlicer/src/slic3r/GUI/GCodeViewer.cpp:2043
#8  0x0000555556975980 in Slic3r::GUI::GCodeViewer::render_toolpaths() const (this=0x5555599418b8) at /media/Gonzales/compile/PrusaSlicer/src/slic3r/GUI/GCodeViewer.cpp:2105
#9  0x000055555696a422 in Slic3r::GUI::GCodeViewer::render() const (this=0x5555599418b8) at /media/Gonzales/compile/PrusaSlicer/src/slic3r/GUI/GCodeViewer.cpp:541
#10 0x0000555556886fde in Slic3r::GUI::GLCanvas3D::_render_gcode() const (this=0x555559941060) at /media/Gonzales/compile/PrusaSlicer/src/slic3r/GUI/GLCanvas3D.cpp:5108
#11 0x0000555556870095 in Slic3r::GUI::GLCanvas3D::render() (this=0x555559941060) at /media/Gonzales/compile/PrusaSlicer/src/slic3r/GUI/GLCanvas3D.cpp:1659
```

It's not consistent though, I also had it crash on `bool GLShaderProgram::set_uniform(const char* name, const std::array<float, 4>& value) const` in GLShader.cpp, invoked by the lambda function `auto render_as_triangles` in GCodeViewer.cpp. So at that time it crashed later. Maybe some threading issues? I'm trying to bisect for the exact commit after 2.2.0 that introduced the problem, but there are quite a lot of commits...

smirgol, 2021-01-16:
It's more difficult to bisect that than I though it would be, as commit ec86d94f026cd304f2544fcf6990cf360f9db235 unnecessarily introduced an `assert(! m_layer_tools.empty() && m_layer_tools.front().has_wipe_tower);` into ToolOrdering.cpp that always crashes PrusaSlicer so the actual issue cannot be tested for. It took quite some commits until this was removed, so there are a lot of commits that cannot be tested because of that. I need to do another bisect and remove this assert manually for each test where this is in the code, that will take some time unfortunately.

smirgol, 2021-01-17:
The commit that introduced the crash for me is this one: 535a27de65decbf0b16f85ff8004dc8935d86775

This fixes the crashes for me:
```
diff --git a/src/slic3r/GUI/Gizmos/GLGizmosManager.cpp b/src/slic3r/GUI/Gizmos/GLGizmosManager.cpp
index e3bb964ad..1c977c7b9 100644
--- a/src/slic3r/GUI/Gizmos/GLGizmosManager.cpp
+++ b/src/slic3r/GUI/Gizmos/GLGizmosManager.cpp
@@ -413,7 +413,8 @@ bool GLGizmosManager::gizmo_event(SLAGizmoEventType action, const Vec2d& mouse_p
 
 ClippingPlane GLGizmosManager::get_clipping_plane() const
 {
-    if (! m_common_gizmos_data
+    if (!m_enabled
+     || ! m_common_gizmos_data
      || ! m_common_gizmos_data->object_clipper()
      || m_common_gizmos_data->object_clipper()->get_position() == 0.)
         return ClippingPlane::ClipsNothing();
```

I'll also create a pull request.

smirgol, 2021-01-18:
I'm still unable to pin down the issue, there are just to many commits and structural changes to test. But I think it's buffer related. When I try to load a gcode file, it will crash when generating the index buffer, or shortly after when it's trying to use the buffer. That's the same when slicing most objects, it will crash in the render lambda function, so something is off there that for some reason works on older systems but not on mine, which is bleeding edge driver wise.

lukasmatena, 2021-01-19:
Just so that you don't think that nobody cares. I'm glad that you started investigating and am watching you. Sadly, I cannot reproduce the problem and although I investigated 535a27d, I did not see how it could break anything.

aburubh, 2021-01-25:
I'm experiencing the same issue here after upgrading to the latest ver. of PrusaSlicer as it crashes crash while slicing a large object with high infill, or multiple small objects which takes time to print, and the issue is consistent and I can reproduce it.
to check it out, you can test with "Core_J_Primo_V1.STL" from https://www.thingiverse.com/thing:4493589 and follow the designer specifications by making the infill 70%.

The same object was working perfectly with the previous ver. of PrusaSlicer, for now I'll reinstall older ver. until the issue is fixed.

tullo-x86, 2021-01-27:
Hi, I'm the reporter of #5889. I found a workaround on my machine, which was to disable Mesa's use of DRI3 by starting PrusaSlicer with `LIBGL_DRI3_DISABLE=true prusa-slicer`. Not sure if it'll work here, but it's worth a shot.

If it does successfully work around the issue, please say so; it could help the devs get to the bottom of it.

aburubh, 2021-01-27:
Thank you for your feed back.
I'm not sure how / where to set LIBGL_DRI3_DISABLE=true, I did try  to do that in the Environment Variables but it did not work for me, I'm still experiencing the same issue.
BTW, I'm using Windows 10.

smirgol, 2021-01-27:
> Thank you for your feed back.
> I'm not sure how / where to set LIBGL_DRI3_DISABLE=true, I did try to do that in the Environment Variables but it did not work for me, I'm still experiencing the same issue.
> BTW, I'm using Windows 10.

That is a setting for Linux, it probably won't do anything on Windows. But it's interesting, that a similar thing seems to happen on Windows too.
I happen to have a virtual machine for a Win7 installation that I do need for Fusion 360, so meanwhile I run PrusaSlicer 2.3. in that one for now to be able to use the latest features. It didn't crash yet - but it's a virtual machine.

On a side note, crashes got even worse for me. I'm now unable to slice even one of the objects mentioned in my initial post. I rarely can slice anything successfully in 2.3. on Linux by now.

smirgol, 2021-01-27:
The issue seems to be fixed on the latest commit d68489c7d7389fc000960b92a8b193b6880c7322. 
My tests before were done on commit d06aa60691025dfa261bca8b247c1e7255296829 from 18th of January.
There have been a couple of changes to files that I suspect to be the cause of the crash, but I haven't pinned down what actually fixed it.

smirgol, 2021-01-27:
Commit ee40ab421a3270b42dfadd08f47259a071177b93 seems to be the one that fixed the issue.

tullo-x86, 2021-01-27:
> Commit [ee40ab4](https://github.com/prusa3d/PrusaSlicer/commit/ee40ab421a3270b42dfadd08f47259a071177b93) seems to be the one that fixed the issue.

Interesting. I reproduced my issue using the `prusa-slicer-git` package from AUR, which pulls down the default branch (`master`), and I built it on Monday (2021-01-25). I wonder how recently that commit was merged to `master`?

smirgol, 2021-01-27:
The commit is from 21.01. while the version in AUR is from 18.01, so it's missing that specific commit. If I'd know how to build an AppImage, I could provide you with my binary.

tullo-x86, 2021-01-28:
I see you aren't familiar with AUR packaging — the `prusa-slicer-git` PKGBUILD in AUR fetches the latest source from GitHub every time it builds. All of the `-git` packages do this. Any non-`-git` package pulls a specific tag or commit, and any `-bin` package downloads binaries and repackages them for an Arch installation.

At any rate, I've found my workaround until this gets fixed. Hopefully what I've reported is enough information to help the devs find the root cause.

smirgol, 2021-02-24:
FWIW: It's still crashing sometimes, even with the latest commits, especially when trying to slice in 0.3 Draft Mode, but at least it works to some degree now.

bubnikv, 2021-09-02:
Is this issue still relevant? Is it reproducible with PrusaSlicer 2.4.1-alpha1?

tullo-x86, 2021-09-14:
It's no longer reproducible in v2.3.3 on my machine:

* Arch Linux
* Zen kernel v5.13.13
* KDE Plasma v21.08.1 (X.org)
* Mesa v21.2.1
* Radeon RX 6800

My guess is that it was a race condition, and Mesa's performance between v20.3.x and v21.0.x managed to mess with PrusaSlicer's expectations.

nikel123, 2021-10-17:
It looks like some bug in mesa AMD radeon drivers and mesa threaded opengl stuff. It doesn't crush on me if I run prusa slicer with the following env var: mesa_glthread=false prusa-slicer
Also narrowed it to prusa slicer  gcode viewer, it crashes if one tries to open the problematic gcode.
As a workaround one can also just generate gcode by choosing export gcode from the menu. In this case gcode viewer isn't triggered and slices don't crash.