prusa3d/PrusaSlicer #5748: Support Enforcer don't create interface layers

Saij, 2021-01-11:
### Version
Version: 2.3.0+win64
Build: PrusaSlicer-2.3.0+win64-202101111315

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 17,110MB

### 3D printer brand / version + firmware version (if known)
Prusa MINI

### Behavior
The support in the middle of the object attached (created by support enforcers) don't have an interface layer. The interface layer should be created at least on layer 134.

[image]

### 3MF File
[chest.zip](https://github.com/prusa3d/PrusaSlicer/files/5798775/chest.zip)

bubnikv, 2021-02-19:
These interfaces will be created if you enable supports everywhere, not just above the print bed.

bubnikv, 2021-03-18:
Duplicate of #1997 #2801 #3831

Newly the support enforcers take precedence over "on build plate only".

[image]

bubnikv, 2021-03-19:
Fixed with 9f09f03228478aa77549efec845717b06e46cbc3

closed 2021-03-19 by bubnikv

ProTubeVR, 2021-03-22:
Just build myself from master,

The support structure and option are updated,
but still having no interface layers generated below 65° overhand (auto or forced supports)
[image]

bubnikv, 2021-07-19:
> The support structure and option are updated, but still having no interface layers generated below 65° overhand (auto or forced supports)

That is a different issue. The reason why the support interfaces are not created there is that at these angles the object is sufficiently suppoted just by the sparse supports.