prusa3d/PrusaSlicer #5749: Crash when using "fill bed with instances"

napter, 2021-01-12:
### Version
2.3.0

### Operating system type + version
Windows 64

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
When using fill bed with instances with various STLs PrusaSlicer crashes. I've attached an STL as an example.

Steps to reproduct:
Import STL
Right click on object, click fill bed with instances
PrusaSlicer simply disappears

I have my arrange distance set at 0.44 - not sure if that contributes at all.

#### Project File (.3MF) where problem occurs
[Cutter - 2021-01-11T153013.119.zip](https://github.com/prusa3d/PrusaSlicer/files/5799082/Cutter.-.2021-01-11T153013.119.zip)

bubnikv, 2021-01-12:
What printer profile do you have selected? Is the crash reproducible?

napter, 2021-01-12:
@bubnikv it happens no matter what printer profile you have selected. I've tried it with my Prusa and with the system Prusa mini profile.

Yes it is reproducible - it happens every time just follow my steps to reproduce.

bubnikv, 2021-01-12:
I wonder whether it may be a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5745

napter, 2021-01-12:
In this case it slices without crashing so it doesn't seem the same - that bug said it crashes when slicing. Another odd thing to note - it will not crash if you set your arrange spacing to 0 and then fill bed with instances. But if arrange depth is larger than 0 and you select fill bed with instances, it crashes. This behavior has happened to me with multiple STLs - there is one sample attached to the issue to reproduce.

lukasmatena, 2021-01-12:
>  it happens every time just follow my steps to reproduce.

It is not this simple, unfortunately. I for example could not reproduce it on my system (Ubuntu 16.04).

bubnikv, 2021-01-12:
So what is the spacing that we should use to reproduce the crash?

napter, 2021-01-12:
@lukasmatena Just to confirm: Did you use my STL? Is your arrange distance greater than 0? (You can try 0.44). The crash is happening on 2.3.0 with the windows 64 build. I can't speak to the ubuntu release.

lukasmatena, 2021-01-12:
@napter Yes, I tried your STL and 0.44 arrange distance. But I fully trust you that you see the crash. I just wanted to say that it may depend on many things. Yes, it could be platform-specific (or video driver specific or whatever). A backtrace would narrow it down, but I'm not sure what is the easiest way to get one on Windows.
What if you run the application with `--sw-renderer` command-line argument? Would it still crash?

tamasmeszaros, 2021-01-12:
I can reproduce the crash on Manjaro Linux. The fix is on the way.

foreachthing, 2021-01-12:
Maybe someone finds this useful:
STL: Cutter from above.
OS: Windows 10 x64
PS: 2.3.0

* Spacing (_NO rotation!_): 
  1. 6 mm: OK.
  2. 1.77 mm: OK.
  3. 0.88 mm: OK.
  4. 0.44 mm: OK.
* Spacing with rotation:
  1. 6 mm: OK.
  2. 1.77 mm: OK.
  3. 0.88 mm: CRASH.
  4. 0.44 mm: CRASH.
  5. 3.1 mm: CRASH.

napter, 2021-04-14:
Any plan to merge this fix? I've been waiting for it since I have struggled to compile builds for windows, but it doesn't seem to be added to PrusaSlicer 2.3.1-rc.

napter, 2021-04-14:
@tamasmeszaros Any plan to merge this fix? I've been waiting for it since I have struggled to compile builds for windows, but it doesn't seem to be added to PrusaSlicer 2.3.1-rc.

bubnikv, 2021-04-15:
2.3.1 is a minimal patch release. Your fix is in the current master, which is the upcoming 2.4.0

commit 64ec3190 (2021-04-15): Fix for arrange crash when geometry has zero length segments

fixes #5749
closed 2021-04-15 by tamasmeszaros