prusa3d/PrusaSlicer #5750: Arrange throws "could not arrange, geometries invalid" error in 2.3.0 but NOT in 2.2.0

napter, 2021-01-12:
### Version
2.3.0

### Operating system type + version
Windows 64

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Clicking arrange with many of my STL files no longer works. It used to work fine in 2.2 but now in 2.3 it gives the error: "Could not arrange model objects! Some geometries may be invalid."

Here is a screenshot of 2.2 and 2.3 side by side - you can see 2.2 arranged nicely after clicking arrange but 2.3 now refuses to do any arranging and gives the error.

[image]

Steps to reproduce
Import the attached STL
Click "arrange"
In 2.3.0 -> get the error. In 2.2.0 -> arrange works, no error

#### Project File (.3MF) where problem occurs
The STL used to demonstrate the issue.
[Cutter - 2021-01-11T153013.119.zip](https://github.com/prusa3d/PrusaSlicer/files/5799096/Cutter.-.2021-01-11T153013.119.zip)

capter987, 2021-01-12:
i have troubleshooter it it dosent occur at 0 object distance but it does at 0.44

lukasmatena, 2022-04-21:
I am able to reproduce it in 2.3.1, but not anymore in 2.4.2-rc. It probably disappeared after one of the improvements added for the 2.4 series. Closing.

closed 2022-04-21 by lukasmatena