prusa3d/PrusaSlicer #5751: Crashes when started from symbolic link

CrashCash, 2021-01-12:
### Version
2.3.0+linux-x64

### Operating system type + version
Debian Testing

### 3D printer brand / version + firmware version (if known)
Original MK3S

### Behavior
Untar into /opt/PrusaSlicer-2.3.0+linux-x64-202101111322
cd /opt
tar -xjvf /storage/3D_printer/prusa/prusaslicer/PrusaSlicer-2.3.0+linux-x64-202101111322.tar.bz2

Create symbolic link
cd /usr/local/bin/
ln -s /opt/PrusaSlicer-2.3.0+linux-x64-202101111322/bin/prusa-slicer

Start it and it crashes
cd
prusa-slicer 
terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
  what():  boost::filesystem::canonical: No such file or directory: "/root/prusa-slicer"
Aborted

If I create /root/prusa-slicer, then it doesn't throw an error, but it doesn't run, either.

If you start it from the subdir, it runs fine:
cd /opt/
cd PrusaSlicer-2.3.0+linux-x64-202101111322/
cd bin/
./prusa-slicer

bubnikv, 2021-01-12:
I was not able to reproduce your issue, but it may be a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5542
We will likely fix it for 2.3.1

bubnikv, 2021-07-30:
Fixed in PrusaSlicer 2.4.0-alpha1.

closed 2021-07-30 by bubnikv