prusa3d/PrusaSlicer #5754: Not apple silicon friendly ?

GitjoPowershell, 2021-01-12:
### Version
2.3.0

### Operating system type + version
MacOs BigSur
### 3D printer brand / version + firmware version (if known)
Voron 
### Behavior
Prusa slicer 2.3.0 release does not run natively on ARM but rather run on Intel Arch translation 

While on GitHub you stand : 
Apple Silicon builds

We are happy to announce that PrusaSlicer can now be compiled for the new Apple Silicon ARM based architecture. Many thanks to @xarbit and @obelisk for their help in making it happen. (#5313, #5316, #5179, #5124, #5322)
The Apple Silicon builds are compiled against wxWidgets 3.1.4, while all our other builds were compiled against wxWidgets 3.1.3, therefore the Apple Silicon builds will require some additional testing.

I'm not able to get it run natively, neither by compiling or running the latest build

Coud you explain how we are supposed to have PS run natively ?

_Is this a new feature request?_
No

xarbit, 2021-01-12:
@GitjoPowershell 
It was clearly stated that "PrusaSlicer can now be compiled for the new Apple Silicon ARM based architecture."
You will have to build and compile PrusaSlicer on you're own for it to run on Apple ARM as there is no automation or CI/DCD and setup for Apple ARM releases.

Following the Wiki will get you a build that runs on Apple ARM, but it is not a apple app package with fancy icon to click on, it is for developers: 
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Mac%20OS.md

GitjoPowershell, 2021-01-12:
@xarbit 

Thanks for your reply, I try to compile on my MBP following instruction on the page you link : 
https://github.com/prusa3d/PrusaSlicer/issues/5663

So either the fancy way or the compile way did not allow to try PS on apple silicon, at least following instructions does not work for me, as a clarification I often build PS from source in the past on my Intel Mac without any issues.

So for me, it lack clear process to test it.

xarbit, 2021-01-12:
@GitjoPowershell 
how did you install homebrew? are you using native homebrew? Please post the error messages your are getting while compiling.

GitjoPowershell, 2021-01-12:
@xarbit 
Brew was installed using : 
/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)" 

Erros messages are linked on the issue #5663 mainly related to recursive paths: 

CMake Error in **/Volumes/Osx/Volumes/Osx**/Users/jo/Downloads/PrusaSlicer-master/deps/CMakeLists.txt:
Cannot find source file:

/Volumes/Osx/Volumes/Osx/Volumes/Osx/Volumes/Osx/Users/jo/Downloads/PrusaSlicer-master/deps/build/CMakeFiles/dep_wxWidgets
Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
.hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .ispc

Cmakes seems to seek for a /Volumes/Osx in /Volumes/Osx rather than using the correct path.

ammulder, 2021-01-13:
The build process worked OK for me on an M1 Mac Mini with Big Sur 11.1, and afterward I was able to launch PrusaSlicer from `src/PrusaSlicer`

The only issues so far are on launch:
```
[2021-01-12 20:16:51.055025] [0x0000000104d0fd40] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 5): image not found
```

And also the title bar says `PrusaSlicer-2.3.0+UNKNOWN based on Slic3r`

Slicing seems to work fine in the couple tests I tried, but I haven't yet printed models sliced with this build.

GitjoPowershell, 2021-01-13:
@ammulder 
Good to know, so seems that issue Is only on my side, did you install brew using the same cmd line that I post above ?

obelisk, 2021-01-13:
@ammulder You'll get that error even if you start a normal build. I think it comes from a piece of hardware support for manipulating 3D models.

ammulder, 2021-01-15:
@GitjoPowershell I installed brew using whatever command was on the front page of the Homebrew Web site

GitjoPowershell, 2021-01-15:
@ammulder 

thanks I try uninstalling brew and reinstalling and still get same behavior with recursive path on cmake 

can’t understand what happen when cmake file is parsed that result in such path

rtyr, 2021-06-09:
Can we close this issue?

rtyr, 2021-07-12:
No response. Native ARM builds are available since 2.3.1. Closing.

BastR, 2021-07-13:
Hi, I'm sorry but I didn't see the notifications.

To answer the question, the problem is still present with the latest build of PrusaSlicer

bubnikv, 2021-07-14:
@BastR
> To answer the question, the problem is still present with the latest build of PrusaSlicer

What problem? We are churning out universal builds, that we test on an M1 laptop all the time.

BastR, 2021-07-14:
@bubnikv Sorry...I made a wrong post, I thought we were on the issue about the cutting function... 

You can indeed close, it works great on apple silicon.

Thanks

closed 2021-07-14 by rtyr