prusa3d/PrusaSlicer #5755: 2.3.0 installer reboots PC

sobieh, 2021-01-12:
### Version
2.3.0

### Operating system type + version
Windows 10 Pro x64

### Behavior
When installer finishes installing the slicer software simply reboots the machine without asking.
There is absolutely no need to reboot the machine when installing a slicer software.

Selected components were just the slicer software and pronterface (no drivers).

rtyr, 2021-01-12:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3914. Known issue, it should be fixed in the new standalone installer.

You can download&try it here:
https://www.prusa3d.cz/prusaslicer/

[image]

bubnikv, 2021-04-14:
@Janys updated the installer to not install vcredist, thus there will be no restart.
Starting with PrusaSlicer-2.3.1-rc, we are adding api-ms-win-crt-runtime-l1-1-0.dll to the PrusaSlicer installation, so there will be no need to install vcredist even if api-ms-win-crt-runtime-l1-1-0.dll is missing on the target system.

closed 2021-04-14 by bubnikv