prusa3d/PrusaSlicer #5757: [Bug] Issue with double-clicking menu bar to maximize window on macOS

MohanMaker, 2021-01-12:
### Version
2.3 (final)

### Operating system type + version
macOS Big Sur 11.1

### Behavior
When I double click on the top menu bar of PrusaSlicer to shrink the size of the window, the window shrinks but the top part of the PrusaSlicer window goes underneath the top macOS menu bar, and you can no longer move around or resize the window (until you quit and restart PrusaSlicer). This does not happen with other macOS apps. See video.

https://user-images.githubusercontent.com/31443287/104316889-231a5080-54ab-11eb-8540-d87fa7c3edbc.mov

MohanMaker, 2021-01-15:
@rtyr Any ideas about this?

bubnikv, 2021-01-15:
We have seen this issue time to time and we have tried to tackle it before. We still did not find a good solution.

closed 2026-09-29 by prusaslicer-bot[bot]