prusa3d/PrusaSlicer #5762: Lost settings and profiles on uppgrade

radimsejk, 2021-01-13:
### Version
2.3.0-beta1 win 64

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Prusa MK3S with MMU2S

### Behavior
Downloaded new version and installed it. I was asked to put in my admin credentials than it finished OK. After first run of program it showed me dialog with new version and features. I have started the update. Update failed and now I have lost my settings, profiles, filaments.

* _Expected Results_
Expected result is to have my profiles back.

neophyl, 2021-01-13:
Beta versions of the software store their profiles in a different place to the release software (or the alphas).  As such its not surprising that you don't have any profiles unless you have previously had a beta version installed.

My question is were you expecting to install a beta version (in which case you should of been aware of the profiles thing as its mentioned in the beta release notes) or did you expect to be installing the new 2.3.0 full release ?    If you were expecting the proper release then how come you have installed a beta ?

foreachthing, 2021-01-13:
This might help, too: https://github.com/prusa3d/PrusaSlicer/issues/3763

radimsejk, 2021-01-13:
>Beta versions of the software store their profiles in a different place to the release software (or the alphas). As such its not surprising that you don't have any profiles unless you have previously had a beta version installed.

First, thank you for answer and help.
So there is a chance my profiles and settings are safe somwhere on my disk? So if I reinstall to latest alpha, Slicer will find my profiles again?

> My question is were you expecting to install a beta version (in which case you should of been aware of the profiles thing as its mentioned in the beta release notes) or did you expect to be [...]

I have been expecting full 2.3.0 release. I have downloaded installer [PrusaSlicer_Win_standalone_2.3.0.exe](https://www.prusa3d.cz/prusaslicer/) and installed.

radimsejk, 2021-01-13:
I have now downloaded 2.3.0 from here and my settings is back. Thank you all for pointing me to the right direction.

closed 2021-01-13 by radimsejk