prusa3d/PrusaSlicer #5763: Feature Request: Print Time for first layer

bbeller123, 2021-01-13:
### Version
_Version 2.3.0 

#Windows 10 64bit 

### 3D printer brand / version + firmware version (if known)
Prusa i3 SKR Turbo
Marlin 2.0.5.3

_Is this a new feature request?_
Yes

Print Time per feature analysis 
New feature request, print time for first layer

bubnikv, 2021-01-13:
Already implemented if you hover mouse cursor over the vertical slider, you
will find the print time estimate to the layer at the left of the thumb.

bbeller123, 2021-01-13:
[image]

Thank you for the advice.

Is there a way to include the first layer print time in the graph?

rtyr, 2021-05-11:
> Is there a way to include the first layer print time in the graph?

The first layer also contains perimeters/infill/etc., so it may be problematic.

rtyr, 2021-08-12:
Implemented. It will be included in the upcoming PrusaSlicer 2.4.0-alpha1. Closing.

[image]